letscontrolit/ESPEasy #1071: move declaration of vars from ESPEasy.ino into header file

uzi18, 2018-03-12:
I want to split declaration into ESPEasy-Globals.h to have only functions there.
@Grovkillen @TD-er is it ok for You?
I will prepare patch/PR

Grovkillen, 2018-03-12:
Sounds like a good plan to me

TD-er, 2018-03-12:
And structs?
Separate .h file EspEasy_types.h ?

uzi18, 2018-03-12:
first will move all to one file, next we cant think if we need to split more

uzi18, 2018-03-12:
all Vars and structs -> .h

v-a-d-e-r, 2018-03-12:
And where should be the advatage of it? The more stuff is external, the more you need to search things. All in one Editor (Arduino) I found more efficient!

TD-er, 2018-03-12:
The main problem of Arduino's .ino files is that you can only do forward declarations in the 'main' file (ESPEasy.ino) This applies also to the numerous structs with data in ESPEasy.ino.
If you place them elsewhere, you cannot use them as parameters or return values for functions.

And having things separated in files describing their function is much easier to find than in one big file.
Especially when you have no idea what's already present, or how it's called, like most developers of plugins. (me included, I spend a lot of time searching)
When looking through the code you'll find lots of code duplication. Some is just cut'n-paste code, but some is just the same functionality with a different name.
So when fixing one bug, there is another left because code isn't shared and almost impossible to find.

I'm not saying everything has to be in .h and .cpp files, but things that need forward declarations in a .h file and the data structs in another sounds like a plan to me.
Then it is clear where to put things and even more important in what order.
There were several issues about people activating defines, which changed the order of constructing objects and thus build errors because the order of forward declarations was incorrect when using those defines. (e.g. enabling the VCC)

That leaves the question open whether or not the .h files should be in the same directory as the .ino files. That's a question asked by @mvdbro a few weeks ago.
I guess moving things into a directory somehow makes them feel "external" and "something not to be changed" or "not to be looked at".
And another thing, when moving structs, you have to make sure the order of members of that struct has to remain the same, because of the way things are stored in the settings files.

v-a-d-e-r, 2018-03-12:
Ok, I agree, that digging in the source code for searching takes a lot of time. I have changed, let's say 30%, of the source code now for my own purposes and made a lot of improvements and fixed also bugs on the way. And, yes, there is a lot of code double and a real nonsense. E.g. the rule parser! So, when it makes it all easier, then do it. And hopefully it is then possible to make the ADC option selectable with a mouse click in the GUI... :-)

uzi18, 2018-03-12:
Ok tommorow I will #PR with first step.

On the other hand such move will fix compilation in eclipse/Sloeber as I need to do it manualy :)

mvdbro, 2018-03-13:
>  Vader: yes, there is a lot of code double and a real nonsense. E.g. the rule parser!

I agree that the rule engine is quite a piece of crappy code. I've only build it to do some automated tests on a standalone ESP unit. Later decided to release the feature also to the public. Might have been a mistake as it is crappy code, severely limited, very picky on syntax and whatever you consider "real nonsense"

It would be nice to have a professional scripting engine within ESPEasy, something like microPython. The good news is that LisaM has already build it.

v-a-d-e-r, 2018-03-13:
I tried only to be gentle not to call it "crappy code." ;-) But yes, it is! My version is with "hobby programmer skills" just 25% of the original size and it seems to work. No issues yet! :-) If microPython is the final solution...? We will see....

uzi18, 2018-03-13:
hey guys, rules are still in improvement state, we have now more air there then before :)

mvdbro, 2018-03-13:
So pls do a pull request for your version so we can all benefiet from better code

uzi18, 2018-03-13:
@mvdbro all is already in mega branch ...

uzi18, 2018-03-13:
Bunch of bugs are fixed in rules, new features like ANDing and ORing conditions are supported, Time comparision are fixed and extended. 
Next I will try to prepare "LET" command to assign value to variable by name (or upgrade taskvalueset by name not number), ELSEIF, maybe condition nesting.

mvdbro, 2018-03-13:
I meant Vader's code.

uzi18, 2018-03-13:
@mvdbro It looks like mission impossible :)

M0ebiu5, 2018-03-13:
@v-a-d-e-r Maybe you could make some pr or at least upload your source? I think it's not fair to talk about nonsense or crappy code,  without showing how to make it better...

uzi18, 2018-03-13:
It is a good idea as opensource project,  show us the code :)

Budman1758, 2018-03-13:
@mvdbro I don't know if you have seen my post about rules processing time but the recent improvements to the "crappy code" made for some rather impressive (imho) improvements in processing times.  Is calling it "lousy code" an improvement?  Methinks going from over 4 full seconds to about 130 milliseconds is rather impressive "crappy code".
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=4997

I ran those numbers. Updated the firmware. Ran those numbers.  Not bad in my opinion.

uzi18, 2018-03-15:
@TD-er @Grovkillen Look at PR #1075,  as I'm blocked in my mega branch and want to push next PR

TD-er, 2018-03-15:
It is merged, so @uzi18 no excuses anymore, start nerding :)

closed 2018-03-15 by TD-er

uzi18, 2018-03-15:
Thx,  I just started without creating branch for my changes so blocked myself :)

TD-er, 2018-03-15:
You can create a new branch from another (earlier) commit and then cherrypick the new commits to it.
That's always my escape when I make the same mistake :)

uzi18, 2018-03-15:
Yes but it is trouble when lots of unstaged changes and some commits already pushed :P

uzi18, 2018-03-15:
I will try now to remember to create branch first...