letscontrolit/ESPEasy #1072: PiDome MQTT does not use unit number as %id% in topic string

pljakobs, 2018-03-12:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Create a PiDome MQTT controller, use default topic string
Configure a unit number != 0
Configure a switch output and name it "state"
observe the topic /hooks/devices/0/SensorData/state
```
mosqitto:sub -t '/hooks/devices/0/SensorData/state'
```
this will show the state of the configured device.
Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
the topic should be '/hooks/devices/12/SensorData/state'

### Actual behavior
See above

### System configuration

Unit | 12
-- | --
Uptime | 0 days 0 hours 17 minutes
Load | 6% (LC=21464)
Free Mem | 18176 (16080 - LoadFromFile)
Boot | Cold boot (0)
Network
Wifi | 802.11N (RSSI -67 dB)
IP / subnet | 192.168.29.48 / 255.255.255.0
GW | 192.168.29.1
Client IP | 192.168.29.46
Serial Port available: | 1 (128 , 0)
STA MAC | EC:FA:BC:13:C3:64
AP MAC | EE:FA:BC:13:C3:64
Firmware
Build | 20000  - Mega (core 2_3_0)
GIT version |  
Plugins | 46 [Normal]
ESP board
ESP Chip ID | 1295204 (0x13C364)
ESP Chip Freq: | 80 MHz
Storage
Flash Chip ID | Vendor: 0x5E Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DOUT
Flash Writes | 9 daily / 9 boot
Sketch Size | 596 kB (276 kB free)

System |  
-- | --
Build | 20000
Git Build | ""
Local time | "1970-00-00 00:00:00"
Unit | 12
Uptime | 18
Free RAM | 18008
Sensors |  
0 |  
tasknr | 1
TaskName | "button"
Type | "Switch input - Switch"
Button | 0
1 |  
tasknr | 2
TaskName | "relay"
Type | "Switch input - Switch"
state | 0

Hardware:
SONOFF basic

Software or git version:
v2.0 branch, fetched from github March 12th 2018

pljakobs, 2018-03-12:
ok, I'd like to change this, the error is in fact, that the default topic string is wrong (including %id% instead of %unit%).

TD-er, 2018-03-12:
So when you change the topic string in the controller settings, things just work?

pljakobs, 2018-03-12:
yes, indeed. 
The culprit is in stringconverter.ino
´´´
void parseSystemVariables(String& s, boolean useURLencode)
{
  parseSpecialCharacters(s, useURLencode);
  if (s.indexOf('%') == -1)
    return; // Nothing to replace

  #if FEATURE_ADC_VCC
    repl(F("%vcc%"), String(vcc), s, useURLencode);
  #endif
  repl(F("%CR%"), F("\r"), s, useURLencode);
  repl(F("%LF%"), F("\n"), s, useURLencode);
  SMART_REPL(F("%ip%"),WiFi.localIP().toString())
  SMART_REPL(F("%rssi%"), String((WiFi.status() == WL_CONNECTED) ? WiFi.RSSI() : 0))
  SMART_REPL(F("%ssid%"), (WiFi.status() == WL_CONNECTED) ? WiFi.SSID() : F("--"))
  SMART_REPL(F("%unit%"), String(Settings.Unit))
´´´
%unit% does make more sense than %id% anyway, I think the best fix would be to update the topic string in _C006.ino. That should also not make too many people angry, since the subscribers will, of course, not see %id% or %unit% but the unit number.

TD-er, 2018-03-12:
I know it is in the stringconverter (I wrote it ;) ).
But before changing a default subscribe topic of a certain controller, I need to make sure it is indeed incorrect.
The "unit" number has a limited range of 0..31
At least when you're using ESPeasy's own Global Sync.

%id% is not part of the system variables, as far as I remember.
See also the [wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy_System_Variables)

pljakobs, 2018-03-12:
in older versions, %id% was either used for the variable index or the controller number, I'm not 100% sure, but the variable index has gone away and the controller number is no longer used for this.
If unit is limited to 0..31, that might be an issue, but then again, I think one could also use %name% in the publish topic

TD-er, 2018-03-12:
What about %idx% ?
That's something a user may assign to a device/plugin/task in ESPeasy.
So you can be even more flexible.

I'm almost convinced %id% is wrong, because it is not available in ESPeasy. But not sure what would be the better choice.

pljakobs, 2018-03-13:
sorry for taking a bit longer.
one important comment first: I don't use the PiDome environment, I use fhem and just use the PiDome mqtt controller out of convenience. So I would not know what the exact requirements and limitations of PiDome are. 
I have a few older (build 147) ESPEasy sensors in the environment that use the same protocol.
Looking at their configuration, I have configured the IDX/Var for every single sensor to be equal to the unit number (probably because I didn't know what i was doing back then).
Doing some testing %id% is the IDX of that sensor, not the unit number of the ESPEasy unit.
In the v2.0 builds, that's no longer available on a per device basis, although the column denoting the activated controller(s) for a device is called CTR(IDX). 
In any case, %id% seems to no longer have any meaning in v2.0 which basically makes all ESPEasy devices post to the same topic for sensors that have the same device name.
As said, not really knowing what PiDome's requirements are, I would nevertheless assume that this is the part of the topic string being used to group sensors (in my case, for example, I have all the temperature values for my heating grouped together with one ID, making it easy to subscribe to the group).
When I consider a system where one EPSEasy has two or more groups of sensors that should be represented by different devices in whatever system subscribes to it, it would be useful to have the old method with a per-device index back. I would probably call it "group" though, since index was always a bit confusing to me.
It has to be noted, though, that this "group" would, in most cases, have to be unique across all ESPEasy installs.
For my purpose, it can be any string, but PiDome might expect a numeral and even put limits on it. 

I was unable to find %idx% in the code, so I assume you are suggesting to bring back a per-sensor index - I'm all for it (again, as said, I would name it %group%.

Everything else is user configurable anyway, it just needs a bit more documentation.

While we're at it, I wonder if we could not create a more flexible MQTT connector that would be configurable, but I guess that's something to just sit down with and go and look at the result.

TD-er, 2018-03-13:
To start with the last sentence.
I've still got some plans to make a virtual (composite) device, which may collect values from different sensors and send them to a coupled controller.
That will give maximum flexibility, since you may create any group you like, or split values. (e.g. SDS011 or PMSx003 dust particle sensor) and send them to any available controller.

The idx value can be set along with any plugin (Device tab => any device you added).
But it may be it is only available when you have a Domoticz controller active, since it was originally meant to be used with Domoticz.
This idx value is present in the event sent to a controller.

pljakobs, 2018-03-13:
I like the idea of a virtual device. It would have to be able to create multiple sensor groups on a single device as well as have a group spread over multiple devices and report through one or more controllers (think a virtual weather device gathering data from a wind sensor on the roof, a rain sensor outside and a number of temperature sensors).
At least for MQTT, I think that an easily be done by using a configurable part of the topic string.

My idea up there was more along the lines of being a little smarter with the topic strings than we are today.
The PiDome connector, as the example that I'm looking at right now, expects the syntax of the subscribe topic to be 
`// topic structure /Home/Floor/Location/device/<systemname>/gpio/16`
with /gpio/ being the command, /16 being the first parameter of the command and the payload being the second (true or false for a gpio, a float for a pwm etc).
I think it would be a lot more elegant if the subscribe template would also use system variables and acutally subscribe to that topic only. Like
`/Home/%unit%/%idx%/%cmd%/%valname%`
That way, the controller could subscribe to that string precisely and would not have to do a whole lot of static string parsing upon receiving a message.
Just a crude thought, thoug, not tested. It should be easy enough to give it a quick try later on, though.

TD-er, 2018-03-13:
You can do that already using rules.
For the missing variables, you still have to type the literal text so the rules still need some adjustment per ESP node.
But it is already possible to do this and also collect several sensor values to send at once.

pljakobs, 2018-03-13:
I know. I came from the angle of "why do we have three Product specific MQTT connectors when where they really differ is mostly the schema".

TD-er, 2018-03-13:
The bulk of the code is now shared.
If you look at the controller code for C001 (Domoticz HTTP) and C002 (Domoticz MQTT), you can see there is now a helper class to create this (quite) specific schema for Domoticz.

And also a lot of MQTT specific code already has been moved to Controller.ino.
It is easier for the end user to select a controller, which may or may not have specific needs for connecting and message formatting.

pljakobs, 2018-03-13:
ok, so perhaps I have just looked at the "wrong" module then ;-)

closed 2019-10-11 by Grovkillen