letscontrolit/ESPEasy #1076: Feature request: Framed Display - show last IP byte instead of SSID

s0170071, 2018-03-13:
I know my SSID all too well- I set it up. Its always the same. Everyday.
But the IP address, especially the last byte is way more interesting...

Budman1758, 2018-03-13:
More useful also.....

s0170071, 2018-03-13:
Yeah, that’s what I intended to say...

TD-er, 2018-03-13:
Should it be selectable, or be one of 3 texts to show on the top?
The hostname is also the same.
And for my personal use, the connected network may change every now and then, since I have multiple accesspoints, each with a name indicating its location.

Budman1758, 2018-03-13:
Possible to make it so you could put any system variable there?

TD-er, 2018-03-13:
It is, but adding things to storage should be done with care.

Budman1758, 2018-03-13:
Actually does not make a huge difference to me.  Just wondering if it was possible.  Just throwing stuff out there.  :)

mattlward, 2018-03-13:
I vote for 3 texts at the top.  I have various AP's and SSID's and always select a backup AP and SSID.  I would love to see the hostname, SSID and last part of the IP address as well!

s0170071, 2018-03-13:
Or maybe replace the "dots" at the bottom with a line of text... e.g.
`1/4       IP:192.168.100.100`
where 1/4 indicates the page thats currently displaying (left justified) and the IP is right justified. 
On small displays omit the first three numbers of the IP and just write` IP ...100`

TD-er, 2018-03-13:
"1/4" sounds also like an option for showing the page.
I will think about it.
Is it also an option to re-use one of the 12 lines as header or footer text?

v-a-d-e-r, 2018-03-13:
I can not understand for what it can be useful to know the page number in a small display??? In a book yes, in a display with 4 (or 12) "pages" NOT..... "Oh yeah, there was a temerature to see on page 5". Very important to know!! *gg* Same crap like the indicator (was the first thing I kicked).... It's not more than wasting valuable space on the display!

TD-er, 2018-03-13:
What indicator do you mean?

v-a-d-e-r, 2018-03-13:
That thing at the bottom of the OLED (framed) version. Pure senseless....

TD-er, 2018-03-13:
The dots you mean?
These were already there, I only fixed them to show only when a page has any data.
The height of that part is the same as the top line.

Given the font sizes, we could use a 20 pixel font height to allow 2 lines below the top line.

v-a-d-e-r, 2018-03-13:
Yeah, that dot stuff. Useless.... Not more like a gimmick!
BTW: I wrote you a PM in the forum about a new bug!?

TD-er, 2018-03-13:
Hmm not visited the forum yet, will have a look.

mattlward, 2018-03-14:
@TD-er, please do not make a 2 line display only display 1 line of live content.  On the 64x48 I only display 1 line at a time due to font size, and I never see the top line of the display with the signal strength.  On a 128x64, I see the top info bar and 2 lines are legible and then we have the line with the dots.  I tend to agree, the dots are like blinking an LED just because you can in an attempt to make something look more sophisticated.

closed 2019-10-11 by Grovkillen