letscontrolit/ESPEasy #1077: Again a problem with the rules% systime% is not executed.

micropet, 2018-03-13:
With the current Mega Branch, the time query is again not executed.
The source from yesterday worked fine.
I had not changed my rules.

I just flashed the version from yesterday - it works.

uzi18, 2018-03-13:
paste rules here if You can :)
But download it from "Advenced/Flash" menu to have exact copy.

micropet, 2018-03-13:
[rules1.txt](https://github.com/letscontrolit/ESPEasy/files/1808218/rules1.txt)

How do I make a debug log?

uzi18, 2018-03-13:
use serial converter or usb connection if it is NodeMcu

micropet, 2018-03-13:
OK,
is not so easy because the MH-Z19 or the PMS7003 hang on the serial port.

TD-er, 2018-03-13:
You can also write the log to a syslog server.

micropet, 2018-03-13:
It was so long ago when I used a syslog server.

You can not have changed so much in such a short time. :)
Can  you not see why it worked yesterday and  today no more?

TD-er, 2018-03-13:
The rules file, was it downloaded by downloading it from the filesystem, or cut and paste from the webinterface?

micropet, 2018-03-13:
It is as desired from the file system.

TD-er, 2018-03-13:
Just checking, since I did not see anything strange in the file. (judging on the special character view of Notepad++)

micropet, 2018-03-13:
It can not be a "real" mistake in it.
It worked for a few weeks.

micropet, 2018-03-13:
And it also works with the version of yesterday.

TD-er, 2018-03-13:
Wellll....
Last weekend there was an issue (which was fixed yesterday) that failed on lines ending with spaces.
A trim() command (to remove spaces at the end of a line) was missing at the right place.
Rules not edited for weeks were all at once not running anymore with that update

uzi18, 2018-03-13:
Do You mean all time comparisions doesn't work?

uzi18, 2018-03-13:
trimm can't do here anything wrong,  maybe parseTemplate is messing?

micropet, 2018-03-13:
Rules not edited for weeks were all at once not running anymore with that update

What is that supposed to mean?
Either a rule is right or it has mistakes.

You can not see that today differently than in the past.

TD-er, 2018-03-13:
There have been some optimizations in the rules handling.
It is now about 10x faster.
But in that process something may have been overlooked.

Grovkillen, 2018-03-14:
@micropet 
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=4032&p=22813&hilit=Tftpd64#p22813

For windows users... sys log server.

micropet, 2018-03-14:
That's great. And so easy.

I have just flashed the latest mega software and written this syslog.
Unfortunately, I am 50km away from home and can not see what the Neopixels are doing.
But I think that they do not work.
[ESP-206 Syslog Not_OK.txt](https://github.com/letscontrolit/ESPEasy/files/1810525/ESP-206.Syslog.Not_OK.txt)

micropet, 2018-03-14:
I'm home again.
The rules are not working.

uzi18, 2018-03-14:
Thanks for logs :+1: 
looks like this works:

```
on Motion#Pir>0 do      // nur tagsueber anzeigen
     if %systime% > 08:00:00 and %systime% < 20:00:00
          gpio,14,1     // Led2 pink ON
     endif
endon

on Motion#Pir=0 do
     gpio,14,0          // Led2 pink OFF
endon

on MH-Z19#PPM do
    if [MH-Z19#PPM]<=700
        event gruen
    endif
    if [MH-Z19#PPM]>=700 and [MH-Z19#PPM]<=1200
        event gelb
    endif
    if [MH-Z19#PPM]>=1200
        event rot
    endif
endon

```

but events gruen,gelb,rot are not working

uzi18, 2018-03-14:
```
<7>ESP-206 EspEasy: ACT  :         event gruen	192.168.0.206	14/03 11:12:10.417	
<7>ESP-206 EspEasy: EVENT:   event gruen	192.168.0.206	14/03 11:12:10.432	
<7>ESP-206 EspEasy: EVENT:   event gruen Processing time:34 milliSeconds	192.168.0.206	14/03 11:12:10.450	

```

micropet, 2018-03-14:
Yes, that's how I see it. The events are no longer running,
But probably because the systime query is not working.

I flashed the version from the day before yesterday again. It works perfectly.

micropet, 2018-03-14:
[ESP-206 Syslog OK.txt](https://github.com/letscontrolit/ESPEasy/files/1812324/ESP-206.Syslog.OK.txt)
Here is the version from 11.03.2018

uzi18, 2018-03-14:
@mvdbro @TD-er @Grovkillen 

Debug log:

```
>event gruen
EVENT: gruen
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 110: gruen do
RuleDebug: 111: if 17:26:39 > 08:00:00 and 17:26:39 < 20:00:00
RuleDebug: 111: neopixelall,0,250,0,0
ACT  :         NeoPixelAll,0,250,0,0   
RuleDebug: 111: else
ACT  :     else
RuleDebug: 111: neopixelall,0,8,0,0
ACT  :         NeoPixelAll,0,8,0,0     
RuleDebug: 111: endif
ACT  :     endif
RuleDebug: 000: endon
RuleDebug Processing:rules2.txt
     flags CMI  parse output:
RuleDebug Processing:rules3.txt 
     flags CMI  parse output:
RuleDebug Processing:rules4.txt 
     flags CMI  parse output:
EVENT: gruen Processing time:51 milliSeconds

Ok
Memtrace
0: lowest: 9824  sendWebPage-> 11000 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10944 sendContentBlocking-> 10960 handle_rules-> 11512 sendContentBlocking-> 11376 sendContentBlocking-> 11400 sendWebPage-> 10856 sendContentBlocking-> 11400 sendContentBlocking-> 9824 
1: lowest: 10856  handle_rules-> 11560 sendHeaderBlocking-> 11544 sendWebPage-> 11000 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10944 sendContentBlocking-> 10960 handle_rules-> 11512 sendContentBlocking-> 11376 sendContentBlocking-> 11288 sendWebPage-> 10856  
2: lowest: 10856  handle_rules-> 11560 sendHeaderBlocking-> 11544 sendWebPage-> 11000 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10896 sendContentBlocking-> 10944 sendContentBlocking-> 10960 handle_rules-> 11512 sendContentBlocking-> 11376 sendContentBlocking-> 11400 sendWebPage-> 10856 
WD   : Uptime 4 ConnectFailures 0 FreeMem 12200
UDP  : Send Sysinfo message
EVENT: Clock#Time=Wed,17:27
RuleDebug Processing:rules1.txt
     flags CMI  parse output:
RuleDebug: 100: gruen do
RuleDebug: 101: if %systime% > 08:00:00 and %systime% < 20:00:00
RuleDebug: 101: neopixelall,0,250,0,0
RuleDebug: 101: else
RuleDebug: 101: neopixelall,0,8,0,0
RuleDebug: 101: endif

```

mvdbro, 2018-03-14:
Tested same here, and it's not good. The PR with the trim() fix was actually making things worse. Last commit should fix it.

mvdbro, 2018-03-14:
Small test:

```
on demo do
  if [Dummy#Value1]=0
   event second
  endif
endon

on second do
  TEST
endon

```
RX: 19:30:13:893  RuleDebug Processing:rules1.txt
RX: 19:30:13:902       flags CMI  parse output:
RX: 19:30:13:907  RuleDebug: 110: demo do
RX: 19:30:13:910  RuleDebug: 111: if 0.00=0
RX: 19:30:13:913  RuleDebug: 111: event second
RX: 19:30:13:915  ACT  : event second
RX: 19:30:13:922  EVENT: second
RX: 19:30:13:926  RuleDebug Processing:rules1.txt
RX: 19:30:13:930       flags CMI  parse output:
RX: 19:30:13:936  RuleDebug: 100: demo do
RX: 19:30:13:938  RuleDebug: 101: if [dummy#value1]=0
RX: 19:30:13:940  RuleDebug: 101: event second
RX: 19:30:13:940  RuleDebug: 101: endif
RX: 19:30:13:942  RuleDebug: 000: endon
RX: 19:30:13:943  RuleDebug: 110: second do
RX: 19:30:13:946  RuleDebug: 111: test
RX: 19:30:13:953  ACT  : TEST
RX: 19:30:13:954  RuleDebug: 000: endon
RX: 19:30:13:959  EVENT: second Processing time:21 milliSeconds
RX: 19:30:13:962  RuleDebug: 111: endif
RX: 19:30:13:969  RuleDebug: 000: endon

micropet, 2018-03-14:
Now it works again.!!

You should put the user settings from the ESPEasy.ino into a userconfig.h or something.

So I have to write each time i flash the personal settings into the ESPEasy.ino File.

TD-er, 2018-03-14:
That's something we could do along with the issue of splitting of global vars/defines.

micropet, 2018-03-14:
Thats fine.

TD-er, 2018-03-14:
Are you building in Windows or Linux?
For Linux you could make a small script to patch the file using sed commands.

micropet, 2018-03-14:
I use Windows.

micropet, 2018-03-14:
I thing the most Users use Windows.

uzi18, 2018-03-14:
I see now lineOrg is used as action, so trimm before copy is real fix :) thx @mvdbro

uzi18, 2018-03-14:
@micropet @TD-er user overrides are better, as usually we change only some settings like:
rulestimers_max, task_max etc.

micropet, 2018-03-14:
Then Gijs has to act. :)

closed 2018-03-14 by mvdbro