letscontrolit/ESPEasy #1082: I want to contribute with a plugin

mstjerna, 2018-03-14:
Hi

I am using the ESP Easy for all my projects an love it. 
The next project I want to do, is a motorized roller-blind with a cheep Chinese stepper-motor.
I see that the Adafruit_Motor_Shield_V2 is already implemented but it's experience...

Motor and driver card is around **$1** from Ebay, AliExpress...
http://robocraft.ru/files/datasheet/28BYJ-48.pdf
http://42bots.com/tutorials/28byj-48-stepper-motor-with-uln2003-driver-and-arduino-uno/

So to get to the point, before I dig in to the coding I want to make sure it would be possible to do this with the ESPEasy as a plugin. Tips and tricks are more than welcome.

**My main concern is that while the motor is spinning, it will be blocking all other processes?** 
Could this work in ESPEasy? (Code is running fine in standalone with the motor)

`
int stepdelay=5;       // delay between stepper steps ( 5 is considered minimun)

void initHardware()
{
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);
  pinMode(14, OUTPUT);
  pinMode(16, OUTPUT);
}

void FW()
{   for (int i=0; i <= 4080; i++){
digitalWrite(16,HIGH);   digitalWrite(14,HIGH);   digitalWrite(12,LOW);   digitalWrite(13,LOW);  delay(stepdelay);
digitalWrite(16,LOW);    digitalWrite(14,HIGH);   digitalWrite(12,HIGH);  digitalWrite(13,LOW);  delay(stepdelay);
digitalWrite(16,LOW);    digitalWrite(14,LOW);    digitalWrite(12,HIGH);  digitalWrite(13,HIGH); delay(stepdelay);
digitalWrite(16,HIGH);   digitalWrite(14,LOW);    digitalWrite(12,LOW);   digitalWrite(13,HIGH);  delay(stepdelay);
}
}
`

Idéas from this source:
http://www.instructables.com/id/Motorized-WiFi-IKEA-Roller-Blind/

Can someone please give a hint if it will work as a plugin to the ESPEasy, the motor will be running for 30-60 seconds...

mstjerna, 2018-03-15:
Hi

I checked the "ESPEasy.ino"
There isn't any type for 4 GPIO selection?
Can this be added?

#define DEVICE_TYPE_SINGLE                  1  // connected through 1 datapin
#define DEVICE_TYPE_DUAL                    2  // connected through 2 datapins
#define DEVICE_TYPE_TRIPLE                  3  // connected through 3 datapins

uzi18, 2018-03-15:
SENSOR_TYPE_QUAD

closed 2018-03-15 by mstjerna

easy1z, 2018-12-22:
did you successful add this to espeasy as a plugin?
I want control a stepper motor and a linear motor same time. 
I have the stepper motor code, want to make it as a plugin.

mstjerna, 2019-01-07:
ESP Easy is way to slow to controll the motor. There are timeouts and the code freezes the motor.
Look at the already supported I2C addon-board with built in clock.
https://www.letscontrolit.com/wiki/index.php/%22Adafruit_Motor_Shield_v2%22

You can also checkout this topic, I have tested it and it works fine.
http://www.letscontrolit.com/forum/viewtopic.php?t=2356

Br
Michael Stjerna

mkhaled7, 2021-03-12:
Hi @mstjerna, 

I am trying to follow this topic to write a custom plugin to control a stepper motor:
http://www.letscontrolit.com/forum/viewtopic.php?t=2356

Not having prior experience, I was wondering if you help me accomplish this? I need some specific instructions on the requirements for the plugin (how to go about writing it) and afterwards how to build it. Eventually I understand I would flash the .bin file using flashesp tool. I would be very grateful for your help!

Best regards,
M.Khaled