letscontrolit/ESPEasy #1083: Flash Writes counter does not increment when the Rules are edited

ghtester, 2018-03-14:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

Not sure if it's a feature or a bug, but I believe it's not correct. 

### Steps to reproduce
How can we trigger this problem?
Edit Rule Set 1

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
Not tested as currently there's no local access to ESP device.

### Expected behavior
Tell us what should happen?
The Flash Writes counter on Tools - Info page should be incremented.

### Actual behavior
Tell us what happens instead?
The Flash Writes counter on Tools - Info page stays the same.

### System configuration

Hardware:
ESP-12E

Software or git version:
mega-20180311

By the way there's a small typo in Tutorial Rules wiki , table (System) events,  System#Boot Example - 'do' command should be at line above.

Thanks a lot to all contributors of this great project!

Grovkillen, 2018-03-14:
Thanks! Updated the wiki :)

ghtester, 2018-03-14:
Wow, I can't believe you are faster than lightning! :-) Thanks for your excellent work! :-)

Grovkillen, 2018-03-14:
I'm trying to make this my daily job... need to get on top then! :smile:

TD-er, 2018-03-14:
It is explicitly not being counted.
I guess to prevent users from loosing their edits when they hit the daily limit and must power cycle the unit to reset that counter.

ghtester, 2018-03-14:
OK, thanks for explanation, that makes sense. Thanks again & kind regards! :-)

closed 2018-03-14 by ghtester