letscontrolit/ESPEasy #1084: Multiple DS18b20, hard to know which are used

Grovkillen, 2018-03-15:
I have some nodes that uses many DS18b20 sensors (it's kinda in the nature of the senors to have many since they are 1-wire). I have to write down each and every ID (Device Address) and pick the one not used. I would like to be able to see which ones are used (task no?) and only pick the unused sensor.

[image]

TD-er, 2018-03-15:
That's a good idea. 
Maybe also add a warning when duplicates are used, similar to the issue to warn when names are already used.
But still we would want to allow it, otherwise it will be hard to switch sensors.

Misiu, 2019-05-23:
@Grovkillen would it be possible to plug new DS18B20 sensor and rescan for sensors?
This way user will be able to attach sensors one by one and assign them properly (using Your method).

uzi18, 2019-05-24:
It looks like new (json) api is disabled by default

Misiu, 2019-07-17:
@Grovkillen is this implemented (I don't have any device near me to test).
About the UI, maybe when the sensor is used You could display a message near the address?
`28-ff-c9-3b-b3-16-5-64 (device name)`
or use 
`<option value="value" style="font-weight: bold;">28-ff-c9-3b-b3-16-5-64</option>` to display bold text for not used values?

uzi18, 2019-07-17:
it looks like we need some core work here - will add it to my TODO list

Misiu, 2019-07-17:
@uzi18 this would be a great improvement.
What about rescan? Will ESP Easy automatically add new DS18B20 sensors to the list?

Misiu, 2019-07-17:
@uzi18 @Grovkillen please take a look at https://github.com/letscontrolit/ESPEasy/issues/1085#issuecomment-482051441. This can be addressed with the same functionality.

uzi18, 2019-07-17:
@Misiu @Grovkillen first things first, need to add something like manager for all dallases first, than we can think about it ...

Misiu, 2019-07-17:
@uzi18 am I right that now we can store only 8 devices (https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P004_Dallas.ino#L78)
Can't we reuse `savedAddress`.
The HTML dropdown is build [here](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P004_Dallas.ino#L102), so maybe we can add `(used)` to option if that address is already used (is in savedAddress?)
Please correct me if I'm wrong.

uzi18, 2019-07-17:
no, we haven't got list of already used sensors and this is only to select one on list

TD-er, 2019-07-17:
The list in the Dallas plugin is populated with addresses found on the bus after performing a scan.
So it is not stored anywhere.

What is stored is the selected address and that is being stored rather inefficient.
I should be possible to store up-to 4 addresses in the same amount of settings bytes.

But @uzi18 is right, before we do a lot of other changes, there should be a central 1wire manager, which would allow to have multiple instances of the same plugin to be active without interacting with eachother (if using the same pin)
Then we can also extend the plugin to support >1 sensor per plugin.

This last new feature also requires some updates in how plugins should handle sending data to a controller. For example Domoticz should split the sensor updates into a single update per sensor and also include an IDX value per sensor. (not sure yet how we should store that extra IDX)

uzi18, 2019-07-17:
how many bytes of ram we can use? sensor id is 6bytes, gpionum 1byte,extra status 1, so i need 8bytes per sensor for manager.
@Grovkillen @Misiu how many dallas you want to connect?

Misiu, 2019-07-17:
@uzi18 I think that 8 DS18B20 is enough (but this limit should be documented). If one would need more he always can add a second physical device) with another set of sensors (Wemos D1 clone can be bought for less than 2$)

TD-er, 2019-07-17:
A plugin cannot handle more than 4 output values per plugin, so there should be no more than 4 Dallas sensors per plugin.
The address is now stored as a byte per value of 32 bit, so that's why I mentioned it can store 4x as many.

uzi18, 2019-07-17:
I want to search bus and store all data internally in manager, task only will fetch data from manager so it will be possible to have dallas task with 4 sensors later. 
Manager need some amount of memory to store all sensors from buses/gpios with statuses and schedule all work in time

TD-er, 2019-07-17:
I think it does make sense to let the manager scan for sensors and then let all instances of the plugin tell the manager whether or not they are interested in the data from a specific sensor. (incl settings like bits resolution)
So the manager does not have to store anything in flash. It will all do it using RAM and be initialized via the plugins.

uzi18, 2019-07-17:
I have use mode like this in my own project but coded it in pure C with static array.
Here we need something like dynamic list of structs, don't know what is better to choose, need to have fast iteration on list in for loop possible.

TD-er, 2019-07-17:
You could use a `std::map` or `std::list`
But I think you will most likely address them from a plugin, so I would suggest to use `std::vector` for this one.
Then you simply must make a function to compute an id from plugin-ID + variable and use that as an index in the std::vector.
A map does somewhat similar. There you have a key (to address the value in the map) and such a key must be unique.
The map will be able to find the element quite fast.
So the key can be something like an uint16 and made by <pluginnr> * 256 + value_nr.

TD-er, 2019-07-17:
Still nothing with these debug flags set and processing a disconnect:

```
209155 : Waiting for WiFi connect
209258 : Waiting for WiFi connect
scandone

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
�U1061 : 

INIT : Booting version:  (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
1062 : INIT : Free RAM:32672
```

Output from a successful boot + connect:
```
6942 : Waiting for WiFi connect
scandone
state: 0 -> 2 (b0)
7838 : Waiting for WiFi connect
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 8
cnt

connected with MikroTik, channel 8
dhcp client start...
ip:192.168.1.75,mask:255.255.255.0,gw:192.168.1.1
7964 : WIFI : Entering processConnect()
8068 : WIFI : Connected! AP: MikroTik (B8:69:F4:9F:21:FA) Ch: 8 Duration: 3786 ms
8070 : WIFI : Entering processGotIP()
8076 : :urn 48
:urd 48, 48, 0
WIFI : DHCP IP: 192.168.1.75 (breadboard-1) GW: 192.168.1.1 SN: 255.255.255.0   duration: 30 ms
8123 : :ur 1
NTP  : NTP replied: delay 22 mSec Accuracy increased by 0.900 seconds
```

With the debug mode enabled, I do get some stack traces with indication of a failed alloc call.

While processing a WiFi disconnect:
```
SYS  : 3360.00
110709 :  Domoticz: Sensortype: 1 idx: 216 values: 3360
:wr 65 0
:wrc 65 65 0
:ack 65
:rn 258
:wr 64 0
:wrc 64 64 0
:ack 64
:rch 258, 356
:c 1, 258, 614
:rch 356, 266
:wr 64 0
:wrc 64 64 0
:ack 64
:rch 622, 266
:c 1, 356, 888
:wr 66 0
:wrc 66 66 0
[... 214 lines cut ...]
3fff2370:  00000000 00000000 ffefeffe feefeffe
3fff2380:  00000000 00000000 00000001 4026db55
3fff2390:  3fffdad0 00000000 3fff23b8 4026dbe4
3fff23a0:  feefeffe feefeffe 3ffe86f8 401017c9
<<<stack<<<

last failed alloc call: 4026F139(620)

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
�U1060 : 

INIT : Booting version:  (ESP82xx Core 2_5_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.1.2 PUYA support)
1061 : INIT : Free RAM:32672
1063 : INIT : Warm boot #198 Last Task: Task Device timer, id: 8 - Restart SPIFFSImpl: allocating 512+240+1400=2152 bytes
SPIFFSImpl: mounting fs @300000, size=fb000, block=2000, page=100
SPIFFSImpl: mount rc=0
Reason: Software Watchdog
```

uzi18, 2019-07-17:
for manager i will try to use std:list first and pass just ids to task

TD-er, 2019-07-17:
That's also a good idea, just only store a reference to the task + value nr in the list of the manager.

You may want to look if the set ID is stored in the task extra settings or not.
If it is stored in the extra settings, you may want to store it in a list managed by the manager, since loading the Extra Task Settings does take some time and should not be done that often.

Misiu, 2019-07-22:
> A plugin cannot handle more than 4 output values per plugin, so there should be no more than 4 Dallas sensors per plugin.
> The address is now stored as a byte per value of 32 bit, so that's why I mentioned it can store 4x as many.

Is this by design? I was looking for a way to log temperature from 6 DS18B20 into SD card. I've recently found [this PR](https://github.com/letscontrolit/ESPEasy/pull/429) and I want to give it a try, but I need two more DS18B20 sensors.
Is there an option to increase output values from 4 to 8? This would allow us to add more DS18B20.

uzi18, 2019-07-22:
@Misiu I can send you 24 task version of espeasy ;)

Misiu, 2019-07-22:
@uzi18 that would be helpful :) Dzięki 😁
I hopefully will build the hardware by the end of the week.

I'm monitoring this thread, so when Your manager will be ready I can test it!

uzi18, 2019-07-22:
@Misiu
http://forum.abc-service.com.pl/viewtopic.php?t=17

Misiu, 2019-07-22:
@uzi18 dzięki :)

uzi18, 2019-07-22:
Beware 24 tasks version use different format of configuration file

TD-er, 2019-07-22:
When in doubt, run `meminfodetail` command and then check the sysinfo page again :)
Just as a reference, this is the memory-layout of the default settings layout
[image]

Misiu, 2019-08-01:
@uzi18 any updates on the manager?

uzi18, 2019-08-01:
time is not on my/our side ;)

Misiu, 2019-08-01:
No worries :) Take Your time

TD-er, 2019-08-01:
> No worries :) Take Your time

Where? ;)
I'm also looking for it.

Misiu, 2019-10-21:
@TD-er I'm returning to the data logger project. Do You think it will be possible to double the amount of sensor a plugin can handle? You wrote that currently there is a limit of four sensors.
I know that I can use multiple devices and store the data using a simple PHP script, but for a simple temperature logger, a Wemos D1, SD Card and six DS18B20 are enough. I want to avoid extra equipment, especially that the sensors and Wemos will be placed in a place that has no wifi.

TD-er, 2019-10-21:
You could add 6x this plugin and all use the same GPIO pin.

Increasing the number of output values per plugin is something that's going to break all existing settings and very likely introduce a lot of errors..

Grovkillen, 2019-10-21:
The original question is still valid though? The idea of using one task with four DS18b20.

TD-er, 2019-10-21:
> The original question is still valid though? The idea of using one task with four DS18b20.

Yep that should be feasible.

Not sure how far @uzi18 has come with the Dallas code to introduce a single 1Wire line manager (by lack of a better term)

Misiu, 2019-10-21:
> You could add 6x this plugin and all use the same GPIO pin.

I didn't know about that. That will solve my problem :) I'm going to check that and hopefully, the data from all sensors will be stored on to SD card.

TD-er, 2019-10-21:
Well try first just to be sure.
For example by having 1 sensor a bit warmer (keep in hand) so you can see if values are mixed up.
I can imagine it may happen another task is interpreting the data from a sensor and thus the measurements are mixed.
I would suspect that if we have this bug, then the 1st sensor in the task list will take priority, so test by warming the 2nd or 3rd (or later) sensor and see if it does cause issues.

Also there are a lot of fake Dallas sensors out there. One of the reported issues with fake ones is that you cannot use them with more than 3 in a chain.  This is not conclusive for all fakes out there, but it is just one of the (many) reported issues which later appeared to be related to fake ones.

tonhuisman, 2023-09-03:
This seems to be resolved, so can be closed.

closed 2023-09-03 by TD-er