letscontrolit/ESPEasy #1085: Get DS18b20 address as variable

Grovkillen, 2018-03-15:
This is a lazy request. But I have many DS18b20 sensors and plan for more. I have to use a logic analyzer or hook a new sensor up to a ESP in order to quickest (?) find its address. I would love to be able to build a DS18b20 scanner using ESP Easy. A display may publish the address and also have it sent to a server using MQTT or HTTP. I can then print it, store it, or whatever. Today I mark my sensor with this information using a simple dymo printer.

TD-er, 2018-03-15:
Similar to the I2C scanner?
Or frequently poll and log when some new sensor is found including timestamp?

Grovkillen, 2018-03-15:
Yup, the later. I guess it's a very narrow band use case but for what it's used for it's a huge help :+1:

Grovkillen, 2018-03-15:
Two UX categories in a row.... I'm on a streak! :)

Misiu, 2019-04-11:
@TD-er having an option to scan 1Wire for new devices would be awesome!
Currently, I use another device to get the address of DS18B20.

Ideal workflow should look like this:
You attach the first DS18B20, assign a name to it, then You attach the second one and ESPEasy detects that it is a new one (not assigned). This would allow for easily adding new sensors and replacing broken ones.

Imagine You want to attach 10 or 20 sensors. Now You must detect the addresses, wrote them down, and remember which sensor goes where. With those changes, the user will be able to put all the sensors in they destination places, attach one by one and assign them individually.

Grovkillen, 2019-10-10:
@TD-er this would be really useful so I'd say this is part of the release too?

TD-er, 2019-10-10:
It is really useful, but I have no idea how much work it will be.

uzi18, 2019-10-10:
Permanent scanning will give much overhead

uzi18, 2019-10-10:
actual espeasy API does not support something like extra button

Misiu, 2020-08-28:
@uzi18 I don't know the insides on ESPEasy, but I imagine this as an extra button on the interface that would call a function that will scan for new DS18B20 sensors.

tonhuisman, 2024-05-11:
It's been a while, but this request has now been implemented, the `owscan,<rx-gpio>[,<tx-gpio>]` command is available for testing. Use a binary from [this GH Action run](https://github.com/letscontrolit/ESPEasy/actions/runs/9046026086).

Please report your findings here, or in the PR (link above this comment).

uzi18, 2024-05-15:
In fact it was already implemented in web interface by @TD-er in https://github.com/letscontrolit/ESPEasy/pull/3367 for dallas sensors.
Now we have also serial feedback from @tonhuisman

closed 2024-05-29 by TD-er