letscontrolit/ESPEasy #1086: Rotary encoder plugin - bug and enhancement

giig1967g, 2018-03-15:
Hi,
I would like to suggest the following enhancement to the above metioned plugin:
1) possibility to override the internal value of the encoder
(see https://www.letscontrolit.com/forum/viewtopic.php?p=25997 and https://www.letscontrolit.com/forum/viewtopic.php?p=25985)
The request will be used to manually override the internal value of the encoder, in order to set the value the knob will starting from.
Just using taskvalueset does not override the internal value but just the task value.

And to ask to fix the following bug:
2) if the following rule exist, it is fired during system boot:
on rotaryencoder#counter do
...
end on

thanks

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:

uzi18, 2018-03-15:
How do You use this plugin? So we will know why you need to set counter.

giig1967g, 2018-03-15:
Hi,
I want to use it a setpoint for a thermostate.

During the rotation of the knob I dispay the value of the heating setpoint on a oled display.
And this works fine.

But if the setpoint changes from outside (mqtt openhab) and I rotate the knob, the value displayed starts from the internal value of the encoder.
Also, the initial value of the encoder should be set to the value read from MQTT Openhab 

Example: 
I have set in openhab the setpoint = 22°
Then i start the ESP unit and (1) read the setpoint from mqtt and (2) store it in the encoder#counter task using taskvalueset. So in the devices page I see encoder#counter = 22.
I then display the setpoint on the OLED. "Set=22°C"
But, when I start moving the knob, the counter starts from 0, disregarding the value set using taskvalueset. Probably because the value set with taskvalueset is not stored in the encoder but just locally, and while rotating the knob the value is read from the encoder itself.

I hope it's clear now.
More examples can be read in the two threads linked above.

giig1967g, 2018-03-15:
Another example is this:
I rotate the knob and the counter increases.
My logic is that if I press the knob the new counter value is stored otherwise after 5 seconds the new value is forgotten and the previous value is restored (like undo).
But if this last action happens, when I move the knob again, the counter starts from the value before the undo and not from the current value.

Example:
I have encoder#counter = 22
I rotate the knob of two steps. In the OLED I have "Setpoint=24C"
Then I press the knob. the value of 24 is sent to Openhab via MQTT.
Then I rotate the knob 4 steps CCW. the display shows "Setpoint = 20°C". I DO NOT press the knob and after 5 seconds the display shows "Setpoint=24°C"
But if I rotate te knob again, the display starts from 20 and not from 24.

TD-er, 2018-03-15:
Is there a way to get the set value from OpenHab? Then you could use MQTTimport to periodically get the value and use that as "current" value of the encoder.

giig1967g, 2018-03-15:
the problem is not getting the value (that is easy) is to set the current value of the encoder.

TD-er, 2018-03-15:
I get that, but I was just thinking that you would not want to store a copy of the set value on the ESP, but rather on a (central) remote location like OpenHAB.
So I guess it would make sense to have 2 events with some selectable delay (selectable in the plugin) to use in rules.
Rotary#up and Rotary#down with some N indicating the number of steps to keep the number of events low.
And just display the collected value from OpenHab and perhaps also the "+n" or "-n" symbols somewhere on the display too.

giig1967g, 2018-03-15:
Actually, I find the way it works now perfect. 
Very fast and reliable.
So I would not change the plugin itself. i am afraid of inserting unecessary delays.

I just need to set the value the counter will start from the next time it will be rotated.

lhurt, 2018-03-17:
I second that. In fact I like the concept in home automation where the software like FHEM or OpenHAB just orchestrates existing autonomous devices that still work even if the automation software is absent (down).
This leverages the WAF (women's acceptence factor) and the overall convenience very much.

@giig1967g : I tried using the encoder with a led strip and the P59 plugin and found out that the rule processing produces a very significant lag. Do you experience the same?

uzi18, 2018-03-17:
See PR #1103 try to test it.

giig1967g, 2018-03-17:
@lhurt: I find it fast using this rule:
OLED is updated in real time. Super fast.

```
on rotary#counter do
  oled,2,1,Set=[rotary#counter]°C
  timerset,1,5
endon

// if button press, update value
on rotarypush#switch do
  publish /ESP4/dummy/nLocalSetTemp1,[rotary#counter]
  timerset,1,0
endon

//if no button presed within 5 seconds, reset to previous value
on rules#timer=1 do
  taskvalueset,2,1,[mqtt#set]
  oled,2,1,Set=[rotary#counter]z
endon
```

giig1967g, 2018-03-20:
Hi,
tested the encwritre command and it works fine.

closed 2018-03-20 by giig1967g

uzi18, 2018-03-21:
@giig1967g thanks

Zaur949, 2022-01-26:
Hello friends
how to use the "encwrite" command.

uzi18, 2022-01-26:
You can assign actual value to encoder position.:
`encwrite,10`

Zaur949, 2022-01-26:
Thank you friend