letscontrolit/ESPEasy #1087: Add "DEVICE_TYPE_QUAD"

mstjerna, 2018-03-15:
Hi

I am building a plugin for a stepper-motor that require 4 GPIO pins to run.
Please add support for:
#define DEVICE_TYPE_QUAD                    4

Br
Michael Stjerna

uzi18, 2018-03-15:
Please explain more what You need,  there is already type quad

TD-er, 2018-03-15:
@uzi18 There is Sensor_type_Quad
See [SensorTypeHelper.ino](https://github.com/letscontrolit/ESPEasy/blob/786ec4714a8b2b4402a7a1e5f3dd29d939ef53bb/src/_CPlugin_SensorTypeHelper.ino)

But that's about the number of output values. Not about the number of GPIO pins to define for a sensor.

mstjerna, 2018-03-15:
@TD-er

I want to add a stepper-motor. It's directly driven on 4-GPIO's through a ULN2003 chip.
The code for running the motor is already working in a standalone program and I want to port it to a customPlugin so everyone can enjoy it in ESPEasy.

This is the input I need from the Device interface of ESP-Easy. And it's the hardest part for me to understand how to achive :-(

4 GPIO Selection. I only found a "DEVICE_TYPE_TRIPLE"
2 Text inputs. PositionTop, PositionBottom
1 Dropdown, Speed with a 5-8 different values
1 checkbox, Use acceleration

uzi18, 2018-03-15:
If we add now DEVICE_TYPE_QUAD we will be in next place backword incompatible.
On the other side looks like simple change.

TD-er, 2018-03-15:
Hmm, such a low-level stepper motor driver is a bit asking for trouble on an ESP8266.
The thing is you should do such continuous loops only while taking the following into account:

1. The ESP8266 really needs some time "alone" to run background tasks to keep WiFi connected and other things. (or else the watchdog will trigger a reset)
2. The responsiveness of the ESP will grind to a halt when running a lot of bitbanging.

Point 1) is done by calling `delay(...)` or `yield()` every now and then (preferably no longer than 20 msec apart)
Point 2) is more of a problem when running for a number of seconds, or running at a high stepper speed.

For text inputs, see the BMx280 plugin, which has some input field for entering a number and there are combo boxes in the OLED_framed plugin. (contrast)

I have to look into the way settings are being stored to see if increasing the number of GPIO pins can be done.

uzi18, 2018-03-15:
Question is how many steps You want to do at once, as You need to consider to split task into small move parts .

mstjerna, 2018-03-15:
Many... Like 5-10k 
But it should be possible with esp and mqtt as in this projekt
http://www.instructables.com/id/Motorized-WiFi-IKEA-Roller-Blind/

TD-er, 2018-03-15:
For such a blind it doesn't really matter if the motion isn't at continuous speed.
If you would want continuous speed and smooth start/stop, it would be best to add some small ATmega in between which you can send some basic commands like "N steps up".
ESPeasy isn't really real-time

mstjerna, 2018-03-15:
I can add delays... Not a problem.
Want to benifit from the ease of config with the EspEasy thats way more important.

Worstcase i will hardcore the gpio in the plugin.

uzi18, 2018-03-15:
You can add custom settings for plugin,  so I think it is not a problem about no device_quad in core.

mstjerna, 2018-03-21:
@uzi18 
I wrote a plugin with hard-coded pins just for testing. 
Used the "PLUGIN_FIFTY_PER_SECOND" but it's way to slow for this solution. I have now put a dedicated esp12e for each rollerblind without the EasyESP and run a custom FW with MQTT. 
This solution is working as expected without any interrupts or crashes from the watchdog...

I will focus my next project on to read a power-meter connected with modbus on the EasyESP software instead. This task should be inline with the purpose of the software.

Thanks for all the comments in this thread. I really appreciate the community feedback.

closed 2018-03-21 by mstjerna