letscontrolit/ESPEasy #1096: LCD issue with Release v2.0-20180316 and %lcltime_am%

mattlward, 2018-03-16:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Using this task set with an I2C 128x64 display.
[image]

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes
### Expected behavior
Tell us what should happen?
I expected date and time + IP address on 2 lines of the display.  I got that but it also turned the output of this line [D1#RealTemp]{D} Outside into "32° Outside" into "32° Outside201"

### Actual behavior
Tell us what happens instead?
Turned the output of this line [D1#RealTemp]{D} Outside into "32° Outside" into "32° Outside201"  It is broken the same way with the "%lcltime_am%" on any empty display line.  Also broken the same on normal or rotated.

### System configuration

Hardware:  Wemos D1, I2C 128x64 display

Software or git version: v2.0-20180316

TD-er, 2018-03-16:
The am/pm notation was changed as requested.
There is now also a time variant without seconds, which will take less space.

TD-er, 2018-03-16:
See https://github.com/letscontrolit/ESPEasy/issues/1057#issuecomment-373603917

TD-er, 2018-03-17:
Do I need to add a %lcltime_am% without the seconds? Or perhaps even remove the seconds entirely, since it probably won't be updated every second?

Grovkillen, 2018-03-17:
No both are needed. If you tend to use it as a clock seconds are not needed, use as a time stamp you want seconds.