letscontrolit/ESPEasy #1098: Problems with WIFI repeater

Basti77, 2018-03-16:
In my home a Fritz Repeater is needed.

Why does the ESP see my SSID twice?
Other devices only see a WIFI network.

When the repeater is in operation, the ping on the Sonoff switch is not stable.

Can it be that the ESP runs in monitor mode?
but that's not good in mesh environments.

TD-er, 2018-03-16:
What Fritz repeater do you have? One with an ethernet socket, like for example the 1750(E) has?
Is it possible for you to connect the repeater to the wired LAN?
That's just a general advice, since using a repeater as a repeater will slow down the WiFi quite a lot.

The thing is that your repeater has to wait for the packet to arrive, buffer it, start sending, etc. So that takes more than twice the time needed to send it directly and thus occupy the ether.

That being said as general advice (against repeating functionality, the Fritz repeaters used with an ethernet cable are fantastic!). Your problem is with the fact that the ESP does "see" the original and the repeater, both with a different MAC address.
It does only try to connect using a SSID, not to a specific MAC address of the access point.
You could try to name the repeater differently (may have to enable "advanced" mode in the Fritz web interface) to be able to distinguish between both. This has the disadvantage that other devices like your mobile also must know about the "new" network and how to connect to it.
The advantage is you have more control over what part of the network to prefer when both are in range.

As far as I know, it is not (yet?) possible to set a specific MAC address of an accesspoint to use.

s0170071, 2018-03-17:
@Basti77: With Fritz boxes as repeater I do also experience wifi disconnects in the exact moment I access the ESP. It will re-connect instantly so that the issue appears as a 3-5sec delay when you access the web pages.
I am not sure who to report the bug to or how / if they can reproduce it....

Maybe someone with an AVM Repeater and the ability to wireshark the ESP traffic could determine if the ESP disconnects intentionally from an AVM point of view.... 
@TD-er: do you know someone with those abilities ?

TD-er, 2018-03-17:
I've setup my network to be able to do Wireshark, but that's via a special switch which allows to mirror all traffic to 1 port connected to my laptop.
The thing is, disconnects/reconnects can only be detected when the reconnect is to a different accesspoint, not when it is the same.
The Fritzbox modem (not repeater) is also capable to do some (undocumented) wireshark on its own, but then you log to a file, copy that, open in Wireshark, etc.
It is possible to add some log/event in ESPeasy to detect if you're connecting to a different accesspoint.

Basti77, 2018-03-17:
Would it not better to monitor the wifi?
kali linux and the nic put in the monitor mode?

uzi18, 2018-03-17:
Repeater sometimes works on 2 channels thats why 2 ssid with same names You could have.

uzi18, 2018-03-17:
Your problem could be resolved if esp can set channel for wifi connection.

TD-er, 2018-03-17:
@Basti77 There are not that many WiFi adapters that allow real promiscuous mode.

@uzi18 Or test MAC address of the accesspoint.

s0170071, 2018-03-17:
@TD-er : I had opened an repeater issue with the Arduino core guys here: https://github.com/esp8266/Arduino/issues/4332 Unfortunately it somehow stalled.
Would you be so kind and have a look at the wireshark capture I posted there, especially at the sequence numbers for seq and  ack of the last packet ? I may be wrong but they seem swapped to me...

TD-er, 2018-03-19:
Could you test with the next build to see if it is working better?
I changed the way how the WiFi connection is being made and it will try to connect to the same BSSID as was for the last successful connect.
Also added some semi-random time shift to prevent all nodes to connect at the same time (e.g. after power outage or temp. offline of the repeater) and thus crashing things.

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2019-08-26 by TD-er