letscontrolit/ESPEasy #1101: REQUEST: create variables for incoming mqtt messages

fragolinux, 2018-03-16:
please implement some way to create variables for incoming mqtt messages, so that they can be used on displays or what else

for example, at work we have a commercial monitoring system which can publish the values of its sensors... i'd like to subscribe to those topics and display them on the oled, or just publish on display something received from node-red, for example, via mqtt

the mqtt-import plugin only works with numbers... thanks in advance

Grovkillen, 2018-03-16:
Somewhat related #812

mvdbro, 2018-03-16:
If you have node-red running, you could use that to translate incoming MQTT messages to OLED commands to MQTT ./cmd topic on the ESP Easy. I don't use Node-Red but i know you can do stuff like that.

fragolinux, 2018-03-16:
@mvdbro how do you know i could use nodered? :)
thanks for the tip, i'll try this way :)

Grovkillen, 2018-10-23:
Since #812 is related I close this one.

closed 2018-10-23 by Grovkillen