letscontrolit/ESPEasy #1104: REQUEST: ESPEasy Mega 2.0 and SDS011 sleep/wake time

Ostrymiecz, 2018-03-17:
Currently ESPEasy Mega 2.0-dev version has an option for SDS dust sensors to make them go sleep for certain amount of time. The sensor wakes up for 30 seconds and then goes to sleep for entered amount of time again and again. Would it be possible to add another text field, but this time for wake time? I would like to extend 30 seconds to 60 or 90, but this value (30 seconds) is hardcoded.

I think that 30 seconds is a little to short for PM 2.5/10 readouts to stabilise 🤔.

TD-er, 2018-03-17:
The 30 seconds is hard-coded into the SDS sensor, when using the "automatic" mode.
You can only set the "off" time, not the "on" time, when storing it in the flash settings of the sensor.

There is an option to put it to sleep and awake manually as described in chapter 4 of the [datasheet](https://cdn.sparkfun.com/assets/parts/1/2/2/7/5/Laser_Dust_Sensor_Control_Protocol_V1.3.pdf)
I've tried that before, but this wasn't as stable as I would want it to be, so I guess the manufacturer's choice of 30 seconds would be a good setting. And it is practical the sensor does the timing :)

Ostrymiecz, 2018-03-17:
OK, thank You.

TD-er, 2018-03-17:
I think it can be closed, right?

Ostrymiecz, 2018-03-17:
Right. I can use relay shield for my Wemos D1 mini and Rules for that.

TD-er, 2018-03-17:
Why would you want to do that? The manufacturer does do the 30 seconds interval anyway.
Even when running continuously.
That was one of the reasons running it with an own defined interval wasn't running like it should. Its measurements were a lot more fluctuating then.

Ostrymiecz, 2018-03-17:
Because I want to stabilise readouts. I was testing SDS011 and readouts stabilises after 40-45 seconds. If I use relay with rules then I can make it work for 60 seconds and then I will switch relay to off state for 30 minutes and so on.

TD-er, 2018-03-17:
OK, that's different from my experience.
I can add the software on/off to the plugin.
But then I'm very curious what caused the strange readings I saw. When enabling again the measurements were higher than before. Just like there was something still present (either counts or some higher concentration left).

Ostrymiecz, 2018-03-17:
Software on/off sounds interesting. How would it work?

TD-er, 2018-03-17:
Via the same serial TX (ESP) => RX (DS011) cable.
The sensor is still able to communicate when in sleep mode. So you can toggle sleep/active mode.
That was my first implementation, but as I said it gave irregular results when testing, so that's why I changed it to "flash settings" to the sensor. That even would give the option to "program" the sensor once and then remove the TX line again to free up the pin for other purposes.

Ostrymiecz, 2018-03-17:
Is it possible to have two options then? So:
1) Current 30 seconds hard-coded in SDS
2) Software on/off

TD-er, 2018-03-17:
Yes it is possible. Have to think about how to present it.
Maybe some values from 30 sec up to like 255 seconds, where 30 seconds is the "flash to sensor once possible to remove TX line" option.

Ostrymiecz, 2018-03-17:
Question if the software on/off will spin the fan continuously for selected amount of time (like 60 seconds)? I'm asking because I wanted to achieve better air flow to the sensor (by making fan work a little longer).

TD-er, 2018-03-17:
Yes. during 'wake' the fan will spin and the internal laser is powered.
During 'sleep', both are off, but the controller is still accessible.

Ostrymiecz, 2018-03-17:
Let me rephrase my question – Let's say that I select 60 seconds working time with new software on/off mode that You will add. Will fan spin entire 60 seconds then or make pause between hard-coded 30 seconds and the rest of time added by software on/off mode?

TD-er, 2018-03-17:
As far as I remember, the fan speeds as long as the sensor is awake.
For the software on/off mode the status in the flash has to be changed to run in "normal" mode again and then put the sensor in 'software sleep" (no write to the flash for those software sleep/wake mode changes)

Ostrymiecz, 2018-03-17:
Sounds good. Can’t wait for that feature then 😊.

Ostrymiecz, 2018-04-13:
Any updates on this @TD-er?

TD-er, 2018-04-13:
Not yet, I've been way too busy with something regarding our houses and upcoming changes in the law, and for ESPeasy with the WiFi issues.

Ostrymiecz, 2018-04-18:
OK 😊

TD-er, 2022-04-21:
Is already fixed

closed 2022-04-21 by TD-er