letscontrolit/ESPEasy #1105: PR: Plugin 181 & 182 from playground to mega

s0170071, 2018-03-17:
I use plugins 181 (STH2x temperature and humidity sensor) and 182 (ISKRA MT681 smart power meter) now for about a year without change. They work with 2.3.0 and 2.4.0 core.

I consider them stable so they should go to mega to get some broader testing and enrich the ESPEasy ecosystem.

Grovkillen, 2018-10-23:
Should these be merged from the playground then? Added to TESTING. @TD-er

tonhuisman, 2023-09-03:
@s0170071 The SHT2x plugin has been migrated already, but about MT681 I'm pretty sure it isn't included in ESPEasy yet. Is it still relevant to migrate that, rather out-dated, plugin? (It will need a proper overhaul before it is up to current code/coding-standards 🤔)