letscontrolit/ESPEasy #1106: Wrong Terminology: Devices "Delay" should be "Interval"

whatsupskip, 2018-03-17:
I believe for reasons of clarity, the field name should be "Interval", rather than "Delay".
It is possible that a sensor reading has both an interval (frequency at which the reading is taken) and a delay (the time between the sensor being turned on and the actual reading being taken).

### System configuration

Hardware:

Software or git version: I think it includes all current versions including mega-20180316

TD-er, 2018-03-17:
Same applies to the terminology used when setting the Sleep mode.

Grovkillen, 2018-03-17:
Totally agree.

s0170071, 2018-03-17:
While we're at it: I think a "thing" listed under devices  should be device, not a task.

Grovkillen, 2018-03-17:
I understand your point @s0170071 but we should not remove that term since it's really deeply established with commands and documentation.

Grovkillen, 2018-03-17:
One device may occupy multiple task numbers.

PR #1230 (2018-04-13): Renamed delay to interval/time for better understanding
Also renamed "delay" to "interval" and moved the factory reset to bottom of page...

Fixes #1106 #1158 #562
closed 2018-04-13 by TD-er