letscontrolit/ESPEasy #1108: Add M-Bus (Meter-Bus) support

tuxmartin, 2018-03-17:
Hi, 
please add **M-Bus** support.

> The Meter-Bus is a standard for reading out meter data from electricity meters, heat meters, gas meters, etc. The M-bus standard deals with both the electrical signals on the M-Bus, and the protocol and data format used in transmissions on the M-Bus. The role of libmbus is to decode/encode M-bus data, and to handle the communication with M-Bus devices.

Please look at:

https://en.wikipedia.org/wiki/Meter-Bus
https://i.ebayimg.com/00/s/MzJYMzI=/z/4-QAAOSw~epZiD6J/$_86.JPG
https://www1.itron.com/local/Spain%20Product%20Portfolio/GA-CYBLEMBUSSTD-01-EN-10-14-GAS.pdf
https://www.mwatechnology.com/products/gsd8-water-meter-wired-mbus-module/

C library https://github.com/rscada/libmbus and hardware for reading data - https://openenergymonitor.org/forum-archive/node/1944.html (message "_Submitted by TrystanLea on Fri, 29/05/2015 - 14:40_").

[image]

Thanks!

TD-er, 2018-03-17:
Do you also have a specific meter in mind? (with datasheet + protocol)

beicnet, 2018-03-18:
1). [MBus C Library + Protocol](https://github.com/rscada/libmbus)
2). [MBus Hardware + Interfacing (PDF)](https://github.com/rscada/libmbus/blob/master/hardware/MBus_USB.pdf)
3). [MBus Mqtt Logger (Arduino)](https://github.com/roarfred/MBusMqttLogger)
4). [The 1). ported as Arduino Library with Example](https://github.com/gcanitezer/Mbus)

mstjerna, 2018-03-19:
I have one :-)

Schneider iEM3150
https://www.schneider-electric.com/en/product/A9MEM3150/iem3150-energy-meter---63-a---modbus/

TD-er, 2018-03-19:
@mstjerna That's MODBUS, not M-bus.
That's for "the other thread" ;)

mstjerna, 2018-03-19:
@TD-er , whops... Deleting it and moving it...

nardev, 2018-06-30:
@tuxmartin any progress about this?

tuxmartin, 2018-06-30:
@nardev no :-(

hakanakh, 2018-12-14:
Hello friends. is there an improvement in this issue? I used the above circuit diagram. I used the USB-TTL PL2303 for the PL2303 circuit. I'm sending data to the M-BUS client, but I can't get a reply. can you help.

tonhuisman, 2023-09-03:
This issue seems no longer relevant because of long non-reply, and can be closed.

TD-er, 2023-09-03:
Nope, still an open issue