letscontrolit/ESPEasy #1111: Timers, Commands not working with P073 plugin

sentinel073, 2018-03-18:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
**1.** Create a rule to execute SendToHTTP when a timer expires
**2.** Create a rule to execute SendToHTTP on Clock#Time
**3.** Configure a 7 segment display (there is no need for a 7 segment display to be actually connected)
**4.** Tick the "enabled" tick-box to enable the display

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
--> yes

### Expected behavior
The SendToHTTP should be executed, when the timer expires and on Clock#Time

### Actual behavior
When the 7 segment is set to enabled, the timer never expires and SentToHTTP is not executed. SendToHTTP is also not executed on Clock#Time.
In order for SendToHTTP to work, the 7 segment must be set to disabled and the unit rebooted.

### System configuration

Hardware: wemos d1 mini pro v1.1.0

Software or git version:
ESPEasy_mega-20180305

uzi18, 2018-03-18:
For me timers does not work at all on current git mega

uzi18, 2018-03-18:
@sentinel073 please test on actual mega build release: Release mega-20180316

TD-er, 2018-03-18:
I've seen an issue lik this before, including things stop working when the 7-segment display plugin was activated.
Something fishy is going on with that plugin.

uzi18, 2018-03-18:
Ok will check without this plugin first

uzi18, 2018-03-18:
But I don't use this plugin

sentinel073, 2018-03-18:
I have just tried with mega-20180316_test_ESP8266_1024. It doesn't work

uzi18, 2018-03-18:
Rebuild without P073 does not fix for me, so problem is somewhere else.

papperone, 2018-03-18:
as far as I know when I develop the plugin P073 has nothing to do with timers so the issue should be located somewhere else...

TD-er, 2018-03-18:
I was just glancing over the source of P073.
For example the tm1637_i2cXXX functions.
If it is I2C, why re-implementing stuff like that on such a basic level, while it is possibly already available in the Wire library?
And is it possible such low-level interaction with pins possibly also used for some other device may cause undefined behavior?
There is guarantee no other (background) process may be executed between these low-level instructions.

uzi18, 2018-03-18:
Plugin 59 blocks, serial commands for sure.

TD-er, 2018-03-18:
Just another thing about P073.
The pointer Plugin_073_7dgt is never initialized to NULL. That may give strange results.

uzi18, 2018-03-18:
ok it is my WIP encoder write :P

uzi18, 2018-03-18:
@sentinel073 @papperone @TD-er success variable is not setted propperly - unconditionaly true if plugin is used, this is a problem.

uzi18, 2018-03-18:
@TD-er PR ready ;)

papperone, 2018-03-18:
@TD-er : sorry about the naming confusion, as I extract some of the code from a library to incorporate into the plugin I leave the original name for my reference but the communication is not I2C :(

TD-er, 2018-03-18:
Ah that makes sense :)

papperone, 2018-03-20:
I tested with the last PR and cannto see difference, the plugin was stable in my test device and it's still stable... I did not use Timers yet, will try later today, maybe who opened the PR can do the test and reply?

papperone, 2018-03-20:
I read carefully the initial request and it seems related to the Clock#Time event; my plugin uses the "PLUGIN_ONCE_A_SECOND" function, maybe it's related to this?

uzi18, 2018-03-20:
I see You don't understand where is the problem. 
If plugin write return success (true)  it blocks interpretation of commands by espeasy core.
So everything works ok but rules/commands not.

TD-er, 2018-03-20:
Can that also happen for other plugins?

uzi18, 2018-03-20:
Yes, they are affected in such  condition but for this plugin it is fixed.

TD-er, 2018-03-20:
I meant, have you checked other plugins for this?

uzi18, 2018-03-20:
Got it on my list

TD-er, 2018-03-20:
Should we make this issue more general about that, or create a new one to keep things clear?

sentinel073, 2018-03-20:
Thanks very much guys! I tried with mega-20180319_test_ESP8266_1024 and it works :-))

uzi18, 2018-03-21:
@TD-er PR for You #1141

Grovkillen, 2018-03-25:
I close this.

closed 2018-03-25 by Grovkillen