letscontrolit/ESPEasy #1115: No output for commands sent via HTTP

es-ka, 2018-03-18:
### Steps to reproduce
send command via HTTP, e.g.: http://myip/control?cmd=Status,GPIO,16

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)?
Yes

### Expected behavior
Some JSON encoded output

### Actual behavior
No output at all

### System configuration

Hardware: ESP8266 Wemos D1 mini, 4 MB

Software or git version: Bin release v2.0-20180310 with testing plugins

Hint: It worked well with v2.0.0-dev12

TD-er, 2018-03-18:
Could you please test it with a more recent version?
I changed it less than a week ago, so it should be in 20180316 build.

es-ka, 2018-03-18:
20180316 works again as expected. Thank you!

closed 2018-03-18 by es-ka