letscontrolit/ESPEasy #1120: Feature request for DDNS service

happytm, 2018-03-18:
2) is it possible to create plugin for espeasy to use DDNS service so it can be accessed from wan network without using static IP ? example below:

[https://github.com/ayushsharma82/EasyDDNS]

Thanks.

Grovkillen, 2018-03-19:
I would leave that to a router.

happytm, 2018-03-19:
Yes router has to be part of it but to identify and access  particular ESP unit from many deployed on LAN we need some software on these units. This will be not very safe in production environment but it will be helpful for experimenting and coordinating with outside parties in real time.This will also require unique port number for each unit to separate them from each other in ESPEasy-Globals.h file.This unique port numbers could be same as last 2 or 3 digits of your chosen fixed IP address in sketch to make it easy to remember if you have too many units.

#define DEFAULT_PORT        8080                // Enter your Server port value in ESPEasy-Globals.h file.

Thanks

happytm, 2018-03-19:
Yes that's a solution but require an extra hardware and some knowledge of linux. I thought my proposed solution was simple enough to do this. It will require only 1 small library and few lines of code in ESPEasy-Globals.h file. Following is the original code example added to that file after removing unnecessary or duplicate code .

/*
######### DuckDNS Update Client for ESP8266 ##########
This Example Explains the Use of EasyDDNS Library with DuckDNS Service
and Checks for New IP Every 10 Seconds.
Author: Ayush Sharma
*/
#include <EasyDDNS.h>
#include <ESP8266WiFi.h>

const char* ssid = "your-ssid";
const char* password = "your-password";

WiFiServer server(80);

void setup() {
Serial.begin(115200);
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
}
Serial.println(WiFi.localIP()); // Print the IP address
server.begin();

EasyDDNS.service("duckdns"); // Enter your DDNS Service Name - "duckdns" / "noip"
EasyDDNS.client("domain","token"); // Enter ddns Domain & Token | Example - "esp.duckdns.org","1234567"
}

void loop() {
EasyDDNS.update(10000); // Check for New Ip Every 10 Seconds.
}``

TD-er, 2018-03-19:
It does allow for more than one DDNS service, see https://github.com/ayushsharma82/EasyDDNS

If other also want it, please add thumbs-up, because we don't have a better rating system.

happytm, 2018-03-19:
Yes  by activating/deactivating just 1 line of code you have 4 different DDNS service providers option.

Oxyandy, 2018-03-20:
Not sure why on an IoT device this is needed, if you search "[DuckDns Updater](https://www.google.com.au/search?q=DuckDns+Updater )" there are options..
[DuckDNS Client](https://github.com/JozefJarosciak/DuckDNSClient) for example.

ghtester, 2018-03-20:
Well, I would also leave the DDNS to router but it would be really nice to have a possibility to use any custom HTTP port for ESP Easy devices ( already discussed here: https://github.com/letscontrolit/ESPEasy/issues/573 ).

TD-er, 2018-03-20:
It could be used as a Plugin/task I guess.

JG24, 2018-03-20:
I am also waiting for this option. Services such as afraid.org or duckdns.org are very easy to use and the possibility of using them would be great.

tparvais, 2018-06-06:
it would be wonderful to have that plugin available !

closed 2019-10-11 by Grovkillen