letscontrolit/ESPEasy #1124: GPIO16 has no pullup - DeepSleep on ESP-01 doesn't work reliably

Micha-he, 2018-03-19:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
Look in ESP8266 Cumminity Wiki [here](https://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations)

DeepSleep is not reach while the ESP (here ESP-01) is powered with >3.2V via power-adapter or USB.
With Vcc ~ 2.52V, DeepSleep works.

Additional info from the forum [here](http://www.letscontrolit.com/forum/viewtopic.php?p=26004#p26004)

Maybe the implemented DeepSleep-function need a external PullUp ?

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
YES

### Expected behavior
Reliable reach the DeepSleep-Mode.

### Actual behavior
DeepSleep doesn't work porperly.

### System configuration

Hardware:
ESP-01

Software or git version:
Release v2.0-20180313

Grovkillen, 2018-10-23:
Is this a real issue that we need to take care of? I've seen users use deep sleep on ESP-01? Open if I'm wrong.

Micha-he, 2018-10-23:
But the ESPEasy-Code can't work stable, if GPIO16 has no pull-up resistor !

From esp8266.com:
```
GPIO0-GPIO15 can be INPUT, OUTPUT, or INPUT_PULLUP.
GPIO16 can be INPUT, OUTPUT, or INPUT_PULLDOWN_16.
```

From ESPEasy misc.ino (look at line with pinMode):
`bool isDeepSleepEnabled()
{
  if (!Settings.deepSleep)
    return false;

  //cancel deep sleep loop by pulling the pin GPIO16(D0) to GND
  //recommended wiring: 3-pin-header with 1=RST, 2=D0, 3=GND
  //                    short 1-2 for normal deep sleep / wakeup loop
  //                    short 2-3 to cancel sleep loop for modifying settings
  pinMode(16,INPUT_PULLUP);
  if (!digitalRead(16))
  {
    return false;
  }
  return true;
}`

Grovkillen, 2018-10-23:
Ok lets open it again

Micha-he, 2018-10-23:
I'll do another test with the current version in the next days and report here.
But the code is illogical for this isDeepSleepEnaled-function. And with my last test's from march, the function is always false with stable 3.3V Vcc.

Grovkillen, 2018-10-23:
You are probably right since the projects I referred to was battery powered.

s0170071, 2018-10-23:
from www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf: 

**Pin32 serves as a RST pin which can be dangled when it is not used. The reset pin is held low
level when the chip is enabled. In order to avoid reset caused by external interference, the lead
is generally required to be short, and no external pull-up resistor is necessary.**

@Micha-he: you don't need a pullup in gpio16, its included in the reset logic.

Grovkillen, 2018-10-23:
@Micha-he please close this one if what @s0170071 is referring to is correct (which it seems to be).

s0170071, 2018-10-23:
And GPIO16 has an internal pull down, there should be a PULLDOWN option. https://github.com/esp8266/Arduino/issues/478
@Micha-he is right, the ESPEasy code should consider that.

s0170071, 2018-10-23:
hehe... maximum confusion I spread ;-)

Grovkillen, 2018-10-23:
Argh, yep. 👍 😄

Micha-he, 2018-10-23:
With a ESP-01 it's complicated to pull-up GPIO16 :-(
And the isDeepSleepEnaled-function should not work properly, when the command 'pinMode(16,INPUT_PULLUP)' not exists/work, or ?

s0170071, 2018-10-23:
No, would not work. 
Enabling INPUT_PULLDOWN may crash the ESP if you connect GPIO16 to RST.