letscontrolit/ESPEasy #1129: Global sync check box gone...

mattlward, 2018-03-19:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Load version v2.0-20180319

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes
### Expected behavior
Tell us what should happen?
I would expect to see the global sync checkbox, it is now gone!

### Actual behavior
Tell us what happens instead?
[image]

I noticed sync stopped working...

### System configuration

Hardware:
Wemos D1 + framed OLED
Software or git version:
v2.0-20180319

TD-er, 2018-03-19:
Its functionality has been moved to a new controller. Forgot the name, but it has espeasy in the name.

Grovkillen, 2018-03-19:
Global sync is turning into a Controller called "ESPEasy P2P Networking"

TD-er beat me to it!"

mattlward, 2018-03-20:
I see it, but... I can't assign an address, a task or connect it to any thing.  I must be missing something.

mattlward, 2018-03-20:
I found it on the desktop, was not clear on mobile where I tried it first.  The little check boxes really do not show up!

TD-er, 2018-03-20:
But still selecting controllers is apparently hard to do in the mobile gui.
Since @doxikus is also working on a new style, that's good to know.

closed 2018-03-22 by mvdbro

vandalon, 2018-05-09:
you are right, timer2 has become obsolete , just did not delete it.

Gpio state 1 is actually state off, not on. it is set on remotely by homeassistant.