letscontrolit/ESPEasy #1130: Unchecking "Global sync" does not switch off UDP broadcasting

kluszczyn, 2018-03-19:
### Steps to reproduce

1. Turn on "Global Sync" on Tools/Advanced page on unit#1
2. Reboot unit#1
3. Send from another unit command "SendTo,1,PWM,12,500" to check if a connection is working properly. In this case, LED connected to GPIO12 should switch on (~50% brightness).
4. Turn off (uncheck) "Global sync" on Tools/Advanced on unit#1
6. Reboot unit#1
5. Send from another unit command "SendTo,1,PWM,12,0" to check connection. LED will turn off and should not.

### Expected behavior
Unit#1 should not react on SendTo command after switching off "Global sync" option

### Actual behavior
Switching off (unchecking) "Global sync" on Tools/Advanced is not turning off UDP communication between units. The unit is still listed on Main Page/Node List

### System configuration

Hardware: tested on ESP12F, Sonoff basic

Software or git version: mega-dev-20180312

Grovkillen, 2018-03-19:
As discussed here: #1129

The global sync is undergoing a complete redo and is thus turned into a controller.

mvdbro, 2018-03-22:
Global sync is only about sensors replicating values. The SendTo command is to send commands to another unit using UDP protocol and it works without global sync. Same for the nodelist. This is by design

closed 2018-03-22 by mvdbro