letscontrolit/ESPEasy #1131: Add %epoch% to the system variables

kluszczyn, 2018-03-19:
I would like to propose the addition of a system variable **%epoch%** [defined as numbers of seconds that have elapsed since midnight 1970-01-01, UTC.](https://en.wikipedia.org/wiki/Unix_time)  

Having system variable with a numeric time value (epoch) would allow sending epoch values to controllers like OpenHAB for further easy manipulating/converting into a desirable format and also for easy use in rules/time calculations.

TD-er, 2018-03-19:
The time value for this, is already present in the code. It is the value returned from the NTP query.
And another user asked it also, but then for logging to Grafana.

TD-er, 2018-03-23:
See #1160
I named it %unixtime% since "epoch" is more like a moment and Unix time is often described as "seconds since Unix Epoch"

kluszczyn, 2018-03-24:
Yep, %unixtime% is self-explanatory name. Thank you for implementation. 

Closing this issue.

closed 2018-03-24 by kluszczyn