letscontrolit/ESPEasy #1135: Current json string for values are un-parsable

Grovkillen, 2018-03-20:
```
{
"tasknr":3,
"TaskName":"sniff",
"Type":"Generic - MQTT Import",
"cmdMQTT":0,
"Value2":0.00,
"Value3":0.00,
"Value4":0.00
}
```

See the example above. The value names are in the json path making it not generically parsable. It should be something like this :

```
{
"tasknr":3,
"TaskName":"sniff",
"Type":"Generic - MQTT Import",
"TaskValues":[
{
"ValueName":"cmdMQTT",
"Value":0
} ,
{
"ValueName":"Value2",
"Value":0.00
},
{
"ValueName":"Value3",
"Value":0.00
},
{
"ValueName":"Value4",
"Value":0.00
}
] 
```

Grovkillen, 2018-03-22:
https://github.com/letscontrolit/ESPEasy/blob/6344de0649c5dfabc69daa2ebd6b1a1ec2731165/src/WebServer.ino#L3545

From this row and onward I suggest this update: 
<script src="https://gist.github.com/Grovkillen/35a126f095e650a0e8071596e4c5b1f0.js"></script>

Grovkillen, 2018-03-22:
It will display a json formated like this:

```
{
"System":{
"Build":20100,
"Git Build":"",
"Local time":"1970-00-00 00:00:00",
"Unit":20,
"Uptime":0,
"Free RAM":15120
},
"Sensors":[
{
"TaskNumber":1,
"Type":"Generic - Dummy Device",
"TaskName":"test",
"TaskValues": [
{"Name":"a",
"Value":2},
{"Name":"b",
"Value":2},
{"Name":"c",
"Value":2},
{"Name":"d",
"Value":2}]
}]
}
```

Which will be correctly formatted and not having the path cluttered with custom names.

TD-er, 2018-03-22:
Well, looking at your GistSnippet, it looks like you're becoming a C++ programmer ;)

One thing, what should we do when no name is present?
Or should we add an extra "valuenr" or something like that?

Grovkillen, 2018-03-22:
:smile: I think I have to. I feel bad leaving everything to you. Yes Number should be added to it since it could be zero length return answer.

Grovkillen, 2018-04-06:
This is now fixed.

closed 2018-04-06 by Grovkillen