letscontrolit/ESPEasy #1136: Core version ID shows additional digit ?

s0170071, 2018-03-20:
my info page shows 

Build | 20100  - Mega (core 54cc4634)

whereas github does not show the last digit  on their website.. (54cc463).
https://github.com/esp8266/Arduino/pull/4536

Grovkillen, 2018-03-20:
Do you have a custom build? [image]

s0170071, 2018-03-20:
jep, I am trying to fix crashes when serial bytes are coming in but are not read. But that’s not the point. I am curious about that last digit. Do we have an extra digit or is github missing one or does it even have a meaning ?

TD-er, 2018-03-20:
Sha IDs are much larger, but often only the first few digits are enough for unique identification

s0170071, 2018-03-20:
shall we match our display to the github display ?

Grovkillen, 2018-10-23:
I close this issue, please reopen if it's a real problem that we need to fix.

closed 2018-10-23 by Grovkillen