letscontrolit/ESPEasy #1137: FeatureRequest: Bosch BME680 with IAQ available

sisamiwe, 2018-03-20:
Hi,

is there a chance to update the BME680 playground plugin so that also the IAQ can be read out?
The libs of BOSCH and DFRobot provide that by integrating the BSEC. The BSEC is a precompiled BOSCH lib, where the calculations for the IAQ are in.

What do you think?

TD-er, 2018-03-20:
As long as the source of this IAQ is available, I don't see any  reason not to add support for this sensor.
Including pre-compiled libraries may be complicated for various reasons, ranging from license to build/link errors in the future.

sisamiwe, 2018-03-20:
A description of how to implement and use the precomiled lib, you can find directly at Bosch [here ](https://www.bosch-sensortec.com/bst/products/all_products/BSEC )or at DFRobot [here](https://github.com/DFRobot/DFRobot_BME680/tree/master/examples).

TD-er, 2018-03-20:
And where is the source? I _really_ would not like to include a pre-compiled lib which may cause issues in the future and thus dev time to support.

sisamiwe, 2018-03-20:
Within [this ](https://github.com/DFRobot/DFRobot_BME680/blob/master/examples/docs_for_iaq/BST-BME680-AN008-45.pdf)Bosch document, the how-to is descibed.

How to get IAQ?
Processing.(Reference in docs_for_iaq\BST-BME680-AN008-45.pdf chapter 2.4, page 15)

Platform
esp8266
1.We need to copy the file docs_for_iaq\esp8266\libalgobsec.a and paste the file into the hardware\esp8266\2.3.0\tools\sdk\lib folder.

2.Find eagle.app.v6.common.ld in your esp8266 SDK folder.

(if you do't modify it before, you can copy docs_for_iaq/esp8266/platform.txt to replace)

The linker file found at hardware\esp8266\2.3.0\tools\sdk\ld\eagle.app.v6.common.ld needs to be modifed. Inserting the line libalgobsec.a:(.literal .text .literal. .text.) after the line libm.a:(.literal .text .literal. .text.).

3.Find platform.txt in your esp8266 SDK folder.

(if you do't modify it before, you can copy docs_for_iaq/esp8266/platform.txt to replace)

Finally, we need to change the linker argument, telling the linker to include BSEC file. This is achieved by adding the argument -lalgobsec to the line compiler.c.elf.libs=-lm -lgcc ... found in hardware\esp8266\2.3.0\platform.txt.

TD-er, 2018-03-20:
The text you quote here does indicate exactly what I meant.
The pre-compiled lib is built for core 2.3.0 No guarantee it will work with 2.4.0, which is what we will probably use for the next release after the upcoming v2.0

So please, please please provide the source of the library.
Working with pre-compiled libraries is not guaranteed to continue working with newer releases.

s0170071, 2018-03-21:
We should think about moving to 2.5.0 directly as 2.4.0 has some nasty issues... Thats why there is 2.4.1 already and maybe they release 2.4.2  shortly due to a memory leak in the WiFiclient....

TD-er, 2018-03-21:
Quite possible we will be moving to some newer version rather soon and that's why I'm reluctant to introduce something that's version dependent.