letscontrolit/ESPEasy #1138: Make Boot reason readable in rule

igrowing, 2018-03-20:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
In Arduino C it's possible to know a boot reason of ESP8266 with command:
ESP.getResetReason()
The list of boot reasons (what causes boot or reboot)
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Esp.cpp#L372

I want to make a rule which on boot acts differently in dependency of boot reason:
In case of watchdogs or exception it should send MQTT message.
In case of sleep awake it should turn on relay.
In case of normal boot it should turn on LED.

### Expected behavior
Tell us what should happen?

### Actual behavior
Tell us what happens instead?

### System configuration

Hardware:

Software or git version:
The issue is open per request:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=5053&p=26203#p26205

dersparkiie, 2020-01-06:
I would also like to use the boot reason for different actions.
Could someone please help?!

tonhuisman, 2023-09-02:
This has been implemented, with some restrictions though. Can be closed.

closed 2023-09-02 by TD-er