letscontrolit/ESPEasy #1139: Rotary encoder bugs

giig1967g, 2018-03-20:
### Steps to reproduce
2 bugs:

1)
Create a rotary encoder device (Name=rotary + ValueName = counter) and add the following rule:
```
on rotary#counter do
...
end on
```

This rule is executed just after System#Boot. It shouldn't be executed at that time.

2)
The initial value is always = 0, disregarding of the Limit value.
In other words, if I set LimitMin=10, I would like the encoder to start from 10 and go up. 

### Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
1) it should not execute that rule
2) It should start from the LimitMin value

### Actual behavior
1) it executes that rule
2) It starts from 0

### System configuration

Hardware:
Wemos D1 Mini

Software or git version:
mega-20180320

Grovkillen, 2018-03-21:
That behavior is described here: https://www.letscontrolit.com/wiki/index.php/Rotary_Encoder

> Limit min: is the lowest value that the counter will go to. Observe that if you have the counter on a value below this limit the counter will climb above this value but not below it once it has reached the min limit.
> 
> Limit max: is the highest value that the counter will go to. As with the min limit if the counter is above this limit it will descend to the max level but never climb over it once reached below it.

So I would not consider it a bug but rather a non-wanted behavior. Thus I file this as a enhancement request. :)

Grovkillen, 2018-03-21:
And even though these two issues are related it is recommended that you split them into two separate topics.

giig1967g, 2018-03-21:
ok. Will open a second issue for the 1st bug.
thanks

closed 2023-09-04 by TD-er