letscontrolit/ESPEasy #1143: fast setting of more than 1 dummy value incorrect

omega4h, 2018-03-21:
I have a dummy in an ESP-device with 4 values. If I set these values separatelyin my browser  with

http://192.168.0.50/control?cmd=event,set_z.var1=4277880
http://192.168.0.50/control?cmd=event,set_z.var2=1144
http://192.168.0.50/control?cmd=event,set_z.var3=1011
http://192.168.0.50/control?cmd=event,set_z.var4=133

there is no problem.

But I want set these values thru my homeautomation (FHEM). With the syntax used in FHEM I’ve this command-block I'll get errors:

set ESPEasy_ESP_G01 raw event,set_z.var1=4277341
set ESPEasy_ESP_G01 raw event,set_z.var2=605
set ESPEasy_ESP_G01 raw event,set_z.var3=524
set ESPEasy_ESP_G01 raw event,set_z.var4=81

The problem: only the last command is successfull. But if I put sleeps (in my tests 3 seconds) between each command) all values will be updated.

I discussed this in FHEM-forum and we think: The problem seems to lie with the event handling of the ESP easy firmware. The 2nd command interrupts the preceeding command, the 3rd interrupts the 2nd and so on. Only the last command will not be interrupted and so the value will be set.
In syslog I can see only the winning command.

System configuration
WEMOS D1 mini

Software or git version:
Tested with:  ESPEasy_v2.0-20180311, ESPEasy_v2.0-20180316, ESPEasy_v2.0-20180319

TD-er, 2018-03-21:
From FHEM point of view, is it possible to set these values all at once?
At first glance, I don't see why there could not be multiple entries all in one request.
It still needs to be implemented, but I was just wondering why it wasn't done like that in the first place.

omega4h, 2018-03-21:
> From FHEM point of view, is it possible to set these values all at once?

Not at once. Also in FHEM these are 4 single commands - only just fast one after the other

TD-er, 2018-03-21:
OK, then it is just a concurrency issue, no new feature :)

TD-er, 2018-03-21:
I have looked into the source and I guess there is room for improvement.

- The `Command.ino` file has a lot of string compares which are executed always, no if-then-else
- The string compares are case-insensitive, better change it to either upper or lower case and perform a simple byte compare, which is much, much faster.
- Better split into a 'decode command-and-convert-to-enum' and one big switch statement.
- Create some queue of events to process and process asynchronous.

uzi18, 2018-03-21:
Why use event here and not taskvalueset cmd?

uzi18, 2018-03-21:
Any log about problems in FHEM?

omega4h, 2018-03-21:
> Why use event here and not taskvalueset cmd?

I'm happy to know a little bit how to use all the possibilities of ESPEasy - and in FHEM I've only this:
raw:
Can be used for own ESP plugins or new ESPEasy commands that are not considered by this module at the moment. Any argument will be sent directly to the ESP.
arguments: raw <cmd> [<arg1>] [<arg2>] [<...>]
example: set <esp> raw myCommand p1 p2 p3

With this command (raw), set_z.var1 I call a rule in ESPEasy like this:
on set_z.var1 do
  TaskValueSet 6,1,%eventvalue% 
endon

> Any log about problems in FHEM?

No, nothing in FHEM and nothing in ESPEasy-Syslog 

In FHEM there's a parameter called 
maxHttpSessions: 
Limit maximal concurrent outgoing http sessions to a single ESP.
Set to 0 to disable this feature. At the moment (ESPEasy R147) it seems to be possible to send 5 "concurrent" requests if nothing else keeps the esp working.
Possible values: 0..9
Default: 3
I set this parameter to 1 but there was no difference.

Only sleep between the commands helps

TD-er, 2018-03-21:
@uzi18 probably meant, why use the event command and not TaskValueSet command?

Something like http://192.168.0.50/control?cmd=event,set_z.var1=4277880
See the [wiki](https://www.letscontrolit.com/wiki/index.php/ESPEasy_Command_Reference)

omega4h, 2018-03-21:
I tested it:
set ESPEasy_ESP_G01 raw TaskValueSet,set_z.var1=4281930
set ESPEasy_ESP_G01 raw TaskValueSet,set_z.var2=1123
set ESPEasy_ESP_G01 raw TaskValueSet,set_z.var3=999
set ESPEasy_ESP_G01 raw TaskValueSet,set_z.var4=124

None of my dummy vars would be set.

Next test with the rigt syntax (I hope):
set ESPEasy_ESP_G01 raw TaskValueSet,6,1,4281930
set ESPEasy_ESP_G01 raw TaskValueSet,6,2,1123
ESPEasy: set ESPEasy_ESP_G01 raw TaskValueSet,6,3,999
ESPEasy: set ESPEasy_ESP_G01 raw TaskValueSet,6,4,124

None of my dummy vars would be set.

Only with event-command and sleep between works

uzi18, 2018-03-22:
PR #1150 ready

susisstrolch, 2018-03-22:
TaskValueSet doesn‘t work „over the wire“. 
It‘s an internal only. 
I‘m also using the event method because of this restriction. 

Sent by mobile device

uzi18, 2018-03-22:
@susisstrolch this PR will resolve this problem

mvdbro, 2018-03-22:
The ESP uses a single buffer for events from the web. I think that's the limitation that you encounter.
Maybe it needs to be changed in such a way that events are being processed during the webrequest but that could be tricky in some cases.

If the ESP had sufficient RAM, we would have had nice queues for both incoming commands/events and outgoing messages from sensors... We had this on the Nodo project, but events where much smaller back then.

uzi18, 2018-03-22:
Ok right, will update it to process event directly.
Now we got fast processing of rules so queue is not needed.

mvdbro, 2018-03-22:
I think we should check current design and risks before we change this. Direct processing would mean that the rule engine will run while background processes like the webserver are being served. I'm not sure it this happens already and if/when it could lead to all sorts of unexpected concurrency issues.

If we are planning to release current dev as V2.0 this may not be good idea atm.

uzi18, 2018-03-22:
It is working already like that in /tools?cmd=xxx

TD-er, 2018-03-22:
I think the commands should get some make-over, which can be done in several steps, without breaking stuff and perhaps stop if there is enough improvement for the next release.

I think the parsing of the strings can be done with minimal effort (+/- 1h programming) and massive speeds improvements for the parsing alone.

The commands can be stored in a vector of strings, before being parsed (using the same C++ object used currently in the web log)
That allows for async processing when things get too busy.
Just call the processing from the loop() when there is something to do and let the commands be stored in this command-queue.
This queue can be as small as 2 items, or more when needed.

uzi18, 2018-03-22:
how do You want to add plugins commands to this vectors table?

TD-er, 2018-03-22:
Like I said, can be done in several steps, but I wanted to make a special 'registry' for finding commands of plugins to make it resolve faster.
And also included in the initial check could be a quick "allowed/not allowed".
I will write a simple PR proposal this evening to make it clear what I mean.
I can explain things better in C++ than in human languages ;)
One PR can tell you more than a thousand words ;)

uzi18, 2018-03-22:
You got me :)

TD-er, 2018-03-22:
@uzi18 I created a PR #1151 to show what I meant.
I think it will reduce the resources used for parsing the commands quite a lot.

Could you please test it?
I'm not going to merge it this evening, since I'm feeling quite sick and I'm off to bed now.
So I'm not sure about my programming abilities right now and it would be nice if a second set of eyes was looking at it.

uzi18, 2018-03-22:
@TD-er,  will try.

uzi18, 2018-03-22:
This is easy part,  I'm curious how to implement it into plugins, as different callback  with return of enum?

TD-er, 2018-03-22:
The plugins is for tomorrow.
This is already a lot less resource hungry and should not change functionality.

uzi18, 2018-03-22:
Ok I will add Your repo to. my local git and pull this to tests,  rest TD-er.

TD-er, 2020-09-24:
This can be closed as it is no longer an issue now we have the event queue and many many other related fixes.

closed 2020-09-24 by TD-er