letscontrolit/ESPEasy #1145: Current Mega Version does not boot on Wemos D1

micropet, 2018-03-21:
If I flash a Wemos D1 with Arduino 1.8.5 over the COM port with the current mega version, it does not start.
Testet with 2 Wemos Boards.

I get:

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset

The version from 15.03.2018 and older works perfectly.

TD-er, 2018-03-21:
And "current" is build 20180321 ?
Are there existing settings present on the flash or just a brand new one?

micropet, 2018-03-21:
Hi Gijs.

Yes 20180321 from this morning. Both Wemos are new, without ext. Hardware.

TD-er, 2018-03-21:
That's very strange, since I explicitly tested on a module with clean flash.
The build from 0316 also doesn't work?
That's the first with the split of the ESPeasy.ino and also quite a lot of merges. (took me all evening to merge everything)
So if that's the first build with issues, then it's a lot of changes to look through.

Which file did you use? dev/normal/test? The complete filename please.

micropet, 2018-03-21:
The version of 15.03.2018 works perfectly.
The version from 19.03.2018 20.00 clock and the version of today (date 20.03.2018 20.00 clock) do not work.
From 03.16. 2018 works too.

"Which file did you use? Dev / normal / test?"

Do you mean which plugins I have activated? All.

TD-er, 2018-03-21:
No, which bin file (we build each morning at 4am, when something has changed) do you use?
If using your own builds, please also try a pre-built one, just to make sure it is an issue in the code.
The main thing changed in the code since the 16th is the way WiFi is handled.

micropet, 2018-03-21:
I always compile myself. That's why I wrote "Arduino 1.8.5".
I get the .bin files and test.

micropet, 2018-03-21:
Do you have a link?

micropet, 2018-03-21:
When I try to upload the file "ESP_Easy_mega-20180321_test_ESP8266_4096.bin" via the web interface I get the message:

Update error: ERROR[11]: Invalid bootstrapping state, reset ESP8266 before updating

TD-er, 2018-03-21:
`The documentation already says that if you upload a sketch through serial, you'll have to reset the board first manually, before OTA will work, ie. upload via serial, reset the board, then use OTA.`
Taken from here: https://github.com/esp8266/Arduino/issues/3915#issuecomment-349611792

micropet, 2018-03-21:
Resetting is not that easy. I am 50 km from home.
I work here on RDP on a computer home. :)

micropet, 2018-03-21:
With the version ESP_Easy_mega-20180321_test_ESP8266_4096.bin
I get:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
⸮U

INIT : Booting version: (custom)
INIT : Warm boot #24
FS   : Mounting...
FS   : Mount successful, used 76555 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:20072
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216_16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0

Exception (2):
epc1=0x3fff5bd4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5bd4 depc=0x00000000

ctx: cont 
sp: 3fff4de0 end: 3fff5260 offset: 01a0

>>>stack>>>
3fff4f80:  3ffeb610 4027c52b 3fff6844 3fff734c  
3fff4f90:  00000000 4026e31b 3ffef9ec 3fff5d4a  
3fff4fa0:  00000000 3fff52a8 000000e0 401004e4  
3fff4fb0:  3fff68a1 3fff7ca4 3fff5d04 401075a4  
3fff4fc0:  3ffeb1b4 3ffeb1b4 3fff67af 401043a2  
3fff4fd0:  3fff5d04 3fff5c34 3fff5d04 4010441e  
3fff4fe0:  0000002e 00019d03 3fff0220 00000000  
3fff4ff0:  3fff08d0 3fff0220 3fff5020 3fff5709  
3fff5000:  3fff56d4 3fff56d4 3fff5d04 40104508  
3fff5010:  40261b8f 3fff5d04 3fff5d4a 40261b98  
3fff5020:  00000008 3fff5050 3ffef9ec 00000012  
3fff5030:  4026fd04 4026fcd8 3ffefa70 3fff5d58  
3fff5040:  3fff5d04 3fff5cf0 3fff56d4 40267fb9  
3fff5050:  005e0001 3ffefb00 00000000 40272378  
3fff5060:  00000018 3fff5cf0 3fff5d04 40268b1c  
3fff5070:  40272131 0000010f 3fff352c 00000000  
3fff5080:  3fff56d8 00000001 00000000 00000002  
3fff5090:  3fff5630 00000004 3fff5cec 402691a0  
3fff50a0:  3fff7ca4 40248288 3fff352c 3fff5cec  
3fff50b0:  00000000 3fff5d04 3fff5d70 40268b5c  
3fff50c0:  3fff56d4 3fff50f0 00000004 3fff5cec  
3fff50d0:  00000016 3fff56d4 3fff5cec 402682ed  
3fff50e0:  3fff56d4 3fff50f0 00000004 40261d58  
3fff50f0:  00000494 00000000 3fff582c 40261d94  
3fff5100:  3fff5cf0 3fff56d4 3fff5c9c 402683be  
3fff5110:  3fff5c7c 3fff3af0 00000010 00000020  
3fff5120:  3fff518c 3fff56d4 3fff5cec 40268597  
3fff5130:  3fff518c 3ffeb2f0 3fff56d4 40268623  
3fff5140:  00000001 00000001 3fff51c0 00000001  
3fff5150:  3fff3c9c 3fff3aec 3fff3c9c 4024e57a  
3fff5160:  00000000 3fff3aec 3fff51b0 401006dc  
3fff5170:  3fff4148 00000215 00000000 40256e84  
3fff5180:  00000000 3fff3aec 3fff3cc8 fb0000e0  
3fff5190:  00000000 3fff3aec 3fff3c9c 3fff4230  
3fff51a0:  00000000 3fff3aec 3fff3c9c 4024faa2  
3fff51b0:  3fff683c 007c008e 4024e3d0 4024e69c

micropet, 2018-03-21:
Ah, if I disconnect the board several times from the voltage, it starts suddenly.

That is not normal.

With the older versions, the Wemos board starts each time.

TD-er, 2018-03-21:
That's with a preset (already existing) setting for the WiFi?
Or after the initial setup via wifi setup?

I really cannot understand what's happening here, since I tested this yesterday very extensively.

micropet, 2018-03-21:
I have entered the WIFI parameters i9n ESPEasy-Globals.h.

With the old versions I am immediately on the network.

micropet, 2018-03-21:
He gets an exception while trying the WIFI connection.

I have now flashed the old and the current version 15 times.
It's always the same. Alt = OK New = not OK

TD-er, 2018-03-21:
But using a pre-compiled version does not work with your changed `ESPEasy-Globals.h`, so it could be the settings were not saved fully since your previous version was crashing.

By the way, you can use a `Custom.h` file of your own and add a simple define to the build process as defined in the `ESPEasy-Globals.h` :
```
#ifdef USE_CUSTOM_H
#include "Custom.h"
#endif
```

I will have a look at it this evening I hope.

micropet, 2018-03-21:
OK

micropet, 2018-03-21:
I delete the wemos with the esptool and then try again.

micropet, 2018-03-21:
So
Wemos reset
Flash deleted with esptool
reset
flashed current version
reset
Result:

RESET: formatting done...
FILE : Saved config.dat
FILE : Saved security.dat
FILE : Saved config.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
NIT : Booting version: (custom)
INIT : Warm boot #289
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:20072
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216_16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0

Exception (2):
epc1=0x3fff5bd4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5bd4 depc=0x00000000

ctx: cont 
sp: 3fff4de0 end: 3fff5260 offset: 01a0

>>>stack>>>
3fff4f80:  3ffeb610 4027c52b 3fff6844 3fff734c  
3fff4f90:  00000000 4026e31b 3ffef9ec 3fff5d4a  
3fff4fa0:  00000000 3fff52a8 000000e0 401004e4  
3fff4fb0:  3fff68a1 3fff7ca4 3fff5d04 401075a4

micropet, 2018-03-21:
No, the second boot.
I did not write the first one - not fast enough  :)

micropet, 2018-03-21:
With:
esptool.py --port COM7 erase_flash
everything should be deleted.

TD-er, 2018-03-21:
The WiFi settings can also be stored inside the ESP

TD-er, 2018-03-21:
Anyone else with the same problems? I really don't see it at the moment and had checked it for a while yesterday evening, but that might have something to do with me getting sick :(

sidey79, 2018-03-21:
I've not the same problem, but a may similar one. The esp starts with his config wifi in ap mode.
I can set there the ssid and the key for my wifi, but the esp will not connect there.

The version from 16.03.2018 works, later versions not.

I compiled the code myself.

TD-er, 2018-03-21:
Could you increase the timeout for WiFi connection in the function `wifiConnectTimeoutReached()`   (in WiFi.ino)
The minimum is now set at 7000 msec
increase it to some ridiculous value like 20000 and see in the log what the connect time actually is.

sidey79, 2018-03-21:
I rised the value to 20000. Connection works.
I then lowered the value back to 7000. It still works. No idea whats goning on now

TD-er, 2018-03-21:
Do you have any idea what the connection time is, reported in the logs?

sidey79, 2018-03-21:
It varies between 1000 and 7016..

Tested four times now:

```
WIFI : WiFi connect attempt took: 6796 ms
WIFI : WiFi connect attempt took: 3935 ms

WD   : Uptime 0 ConnectFailures 0 FreeMem 15640
WIFI : Connecting xxxx attempt #0
WIFI : Could not connect to AP!
WD   : Uptime 1 ConnectFailures 0 FreeMem 14312
WIFI : Connecting xxxx attempt #0
WIFI : WiFi connect attempt took: 2916 ms
```

One time i got it not connecting. I had this problem with two devices. Did you fix something yesterday?

TD-er, 2018-03-21:
Yep, yesterday I fixed an issue when connecting via the initial web setup.
That's a special case, I did not check for correctly.

I just added a PR, which should make the checks a little more robust and perhaps also some rather strange thing I saw in the documentation.

> Meaning of parameters is as follows: 
> 
> * ssid - a character string containing the SSID of Access Point we would like to connect to, may have up to 32 characters 
> * password to the access point, a character string that should be minimum 8 characters long and not longer than 64 characters 
> * channel of AP, if we like to operate using specific channel, otherwise this parameter may be omitted 
> * bssid - mac address of AP, this parameter is also optional 
> * connect - a boolean parameter that if set to false, will instruct module just to save the other parameters without actually establishing connection to the access point

Especially the minimum length for the SSID looks a bit strange.
And given the initial log mentions "SMC" as SSID, I just made a String object from it with the max lengths reserved.

sylvain-lp, 2018-03-21:
Hello,
just to say that I was having the exact same issues on my Wemos D1 mini.
I used latest binaries (4MB builds, plugins_build=normal) from March 19,20 & 21.
I could connect to ESP_EASY_0 AP and set my SSID & password. But when I had the "Wifi Setup Wizard" countdown from 20 to 0 it could never connect to my home WiFi ("back to setup" button only).
I have now compiled & uploaded today's version (not your latest commit from minutes ago). with the timeout set to 20000. I was able to successfully set the ESP8266 to connect to my home wifi. 
All is fine now. thanks for the tips. I was getting crazy checking cables, power, erase_flash, etc.

TD-er, 2018-03-21:
Could you please check the actual times needed to connect to the WiFi?

Maybe the "setup" phase should also do the more 'relaxed' connection procedure, since the initial connect may take a lot longer now with the accesspoint still active.

sylvain-lp, 2018-03-22:
Not sure where to check WiFi connect time. I have set "System Log: Info", then "WiFi Connect".
Here is what's in the log:
2190181 : WIFI : WiFi connect attempt took: 2135312 ms
2342135 : WIFI : WiFi connect attempt took: 2287265 ms
2342168 : WIFI : Connected! IP: 192.168.1.31 (esp8266-4_4) AP: <my_router>
2342295 : WD : Uptime 39 ConnectFailures 0 FreeMem 16336

sylvain-lp, 2018-03-22:
You're right. It gets better after another cold reboot:

5679 : WIFI : WiFi connect attempt took: 4110 ms
93611 : WIFI : WiFi connect attempt took: 92043 ms
100006 : WIFI : WiFi connect attempt took: 98438 ms
105017 : WIFI : WiFi connect attempt took: 103449 ms
109299 : WIFI : WiFi connect attempt took: 107731 ms

micropet, 2018-03-22:
Good Morning.
Version does not compile with Arduino 1.8.5:

D:\arduino-1.8.3\meine\ESPEasy-mega-03-22\ESPEasy\Wifi.ino: In function 'bool wifiConnectTimeoutReached()':
Wifi:209: error: 'DEFAULT_WIFI_CONNECTION_TIMEOUT' was not declared in this scope
   return timeOutReached(wifi_connect_timer + DEFAULT_WIFI_CONNECTION_TIMEOUT + randomOffset_in_sec);

                                              ^

exit status 1
'DEFAULT_WIFI_CONNECTION_TIMEOUT' was not declared in this scope

micropet, 2018-03-22:
good Morning

It looks good. I just flashed two other wemos. Both immediately went to the network.
The other two Wemos that I used yesterday, I can only test this evening.

micropet, 2018-03-22:
Is not alright yet.
2 Wemos start immediately, the other two only after multiple ON / OFF switch.

INIT : Booting version: (custom)
INIT : Warm boot #131
FS   : Mounting...
FS   : Mount successful, used 76555 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:19824
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216-16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0

Exception (2):
epc1=0x3fff5ccc epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5ccc depc=0x00000000

ctx: cont 
sp: 3fff4ee0 end: 3fff5360 offset: 01a0

>>>stack>>>
3fff5080:  00000000 4026e56f 3ffefaec 3fff6610  
3fff5090:  3ffeb2b8 0002b100 3fff5f48 3fff5e42  
3fff50a0:  00000000 3fff53a4 000000e0 401004e4  
3fff50b0:  3ffeb2b8 3ffeb2b8 3fff5dfc 401075a4  
3fff50c0:  3fff01f0 4026ff68 402725cc 401043a2  
3fff50d0:  3fff5dfc 3fff5d2c 3fff5dfc 4010441e  
3fff50e0:  0000002e 3fff0320 3fff5110 40261cac  
3fff50f0:  00000000 00010cf5 40271f5f 3fff5801  
3fff5100:  3fff57cc 3fff57cc 3fff5dfc 40104508  
3fff5110:  40261ddf 3fff5dfc 3fff5e42 40261de8  
3fff5120:  00000008 3fff5150 3ffefb70 00000012  
3fff5130:  4026fef7 3ffefaec 00000012 3fff5e50  
3fff5140:  3fff5dfc 3fff5de8 3fff57cc 4026820d  
3fff5150:  005e0001 3ffefb00 00000002 00000000  
3fff5160:  00000018 3fff5de8 3fff5dfc 40268d70  
3fff5170:  40285662 3ffe85dd 3fff6934 00000000  
3fff5180:  3fff57d0 00000001 00000000 00000002  
3fff5190:  3fff572c 00000004 3fff5de4 402693f4  
3fff51a0:  40248571 4024854d 3fff696c 3fff5de4  
3fff51b0:  00000000 3fff5dfc 3fff5e68 40268db0  
3fff51c0:  3fff57cc 3fff51f0 00000004 3fff5de4  
3fff51d0:  00000016 3fff57cc 3fff5de4 40268541  
3fff51e0:  3fff57cc 3fff51f0 00000004 40261fa8  
3fff51f0:  00000494 00000000 3fff5924 40261fe4  
3fff5200:  3fff5de8 3fff57cc 3fff5d94 40268612  
3fff5210:  3fff5d74 3fff3bf0 00000010 00000020  
3fff5220:  3fff528c 3fff57cc 3fff5de4 402687eb  
3fff5230:  3fff528c 3ffeb3f4 3fff57cc 40268877  
3fff5240:  00000001 00000001 3fff52c0 00000001  
3fff5250:  3fff3d9c 3fff3bec 3fff3d9c 4024e822  
3fff5260:  00000000 3fff3bec 3fff52b0 401006dc  
3fff5270:  3fff424c 00000243 00000000 402570e0  
3fff5280:  00000000 3fff3bec 3fff3dc8 fb0000e0  
3fff5290:  00000000 3fff3bec 3fff3d9c 3fff4334  
3fff52a0:  00000000 3fff3bec 3fff3d9c 4024fd4a  
3fff52b0:  3fff6aa4 007c008e 4024e678 4024e944  
3fff52c0:  00000000 00000000 3fff3d08 40258518  
3fff52d0:  00000000 3fff3d9c 3fff3d08 4024e22d  
3fff52e0:  3fff5cc4 00000000 4024d9cc 40258ae8  
3fff52f0:  3fff1c38 00000000 3fff3d08 40215953  
3fff5300:  3fff6aa4 00000000 402084e8 40208384  
3fff5310:  3fff1c38 0000000c 3fff315c 4023cd1e  
3fff5320:  00000000 00000000 00000000 3fff68cc  
3fff5330:  0000002f 00000015 feefeffe feefeffe  
3fff5340:  3fffdad0 00000000 3fff432e 40257328  
3fff5350:  feefeffe feefeffe 3fff4340 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
⸮U

INIT : Booting version: (custom)
INIT : Warm boot #132
FS   : Mounting...
FS   : Mount successful, used 76555 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:19824
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216-16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0
OTA  : Arduino OTA enabled on port 8266
EVENT: System#Boot
ACT  : NeoPixelAll,0,0,0,0
ACT  : Publish ESP-216/IP,0.0.0.0
ACT  : timerSet,1,60
ACT  : 
WD   : Uptime 0 ConnectFailures 0 FreeMem 16488
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
WIFI : WiFi connect attempt took: 4775 ms
EVENT: Time#Initialized
Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
EVENT: WiFi#ChangedAccesspoint
EVENT: WiFi#Connected
WIFI : Connected! IP: 192.168.0.216 (ESP-216-16) AP: SMC AP BSSID: C8:0E:14:05:BC:C6
EVENT: Clock#Time=Thu,16:48
MQTT : Intentional reconnect
MQTT : Connected to broker with client ID: ESPClient_68:C6:3A:A5:F1:9E
Subscribed to: ESP-216/#
EVENT: MQTT#Connected
WD   : Uptime 1 ConnectFailures 0 FreeMem 14856
EVENT: Clock#Time=Thu,16:49
WD   : Uptime 1 ConnectFailures 0 FreeMem 14840
EVENT: Rules#Timer=1
ACT  : Publish ESP-216/IP,192.168.0.216
ACT  : Publish ESP-216/Time,16:49:14
ACT  : Publish ESP-216/Uptime,1
ACT  : Publish ESP-216/RSSI,-70
ACT  : timerSet,1,60

TD-er, 2018-03-22:
I really don't get it why you get this crash almost immediately after the first connection attempt.
Could you please add a few log lines in the flow after this last log line to see how far it actually gets?
That crash log of yours was what triggered me to copy the password and ssid to strings and work with those.

Can you also (of this Wemos) set the WiFi credentials again from the web interface, to make sure it is being stored like it should.

micropet, 2018-03-22:
I have re-set the WiFi credentials via the web interface.
Flashed again.
Power off/on
Then I had to press the reset button about 30 times until the Wemos started.

I did not understand that:
Could you please add a few log lines in the flow after this last log line to see how far it actually gets?

TD-er, 2018-03-22:
You were the one building your own firmware-builds? Or am I mixing up people now?

I meant this part of the code:
https://github.com/letscontrolit/ESPEasy/blob/940b65e86a427fa9d10635091fc5e0f283fe9ed2/src/Wifi.ino#L245-L282

You can see the last part of the log you posted here, right before the crash.
In the lines following that, there are numerous places where things can go wrong.
Like the `WiFi.begin()` statements. Does it even get there?
Does it see the BSSID not being 0 at the first element? (should be 0 at first boot)
You could set the BSSID where it is defined to the BSSID of your own accesspoint. In my experience that makes it connect a bit faster (not as much as I hoped)

TD-er, 2018-03-22:
Just a check... do you have your IP defined static? Something similar is reported here: https://github.com/esp8266/Arduino/issues/1809#issuecomment-311299382

micropet, 2018-03-22:
No DHCP on all ESPs

micropet, 2018-03-22:
Yes, I compile myself.

The fact is: The version of 19.03.2018 runs immediately.
I do not think the problem lies with me.
My other 30 Wemos boards with ESPEasy and older software are also running smoothly.
I have not changed anything on my network.

TD-er, 2018-03-22:
You reported earlier that the 20180319 version did not work.
I did change a lot based on that information.

So please, please please..... let's take one step back and get the information straight, since I'm loosing my sanity on this issue. (and sleep)

What builds work? Work on all? 
Please also note the last HEX digit of the MAC address of the modules not working.
You're not using DHCP, but static IP.
What versions of files are you using? How are the nodes configured? (pre-compiled, via webinterface)
How many characters are used in the password? Any special characters?
What plugins are being used?

micropet, 2018-03-22:
Oh, did I?
The version from 19.03.2018 and older run perfectly on all Modules.

I have tested the current version on the modules with this MAC - they are not running .: 68: c6: 3a: a5: f1: 9e and 5c: cf: 7f: d9: 1c: 96
Everything is OK with the version from 19.03.2018.

As I wrote above: all modules use DHCP

The password has 14 letters
on ! is in

All plugins are loaded on all 30 modules.
All modules are almost identical.

The modules are configured from the settings in ESPEasy-Globals.h, formerly ESPEasy.ino.

TD-er, 2018-03-22:
The logs you post sometimes display a boot.mode.
This boot mode is described [here](http://arduino-esp8266.readthedocs.io/en/latest/boards.html#boot-mode)
If I understood correctly, the `boot mode:(3,6)` means it is in flash mode.

On the same page there is a schematic about "improved stability", which in fact tries to force the pins to start the sketch:
[image]

Could you, on the problematic version, please look for this boot mode message in the logs and perhaps add a pull-up resistor to GPIO0? 
Or inspect the solderings of that flash-switch and resistor on the Wemos?

micropet, 2018-03-22:
According to the wiring diagram, GPIO0 has a 10k pull-up.

micropet, 2018-03-22:
I do not believe in a hardware problem. At the two new Wemos nothing is connected.

I mean it is clearly due to the software ..

TD-er, 2018-03-22:
Yes I know, otherwise it will never boot the flashed sketch.
And there is probably also a switch to pull the GPIO0 to GND.

But given the boot mode message in your logs, it is sometimes in 'flash' mode.
So either GPIO0 is not pulled up enough, or it is pulled down too much.
Thus the resistor is not soldered well enough, or the boot/flash switch is pulling it down to GND.

micropet, 2018-03-22:
Should I solder in an additional 10k resistor?

TD-er, 2018-03-22:
Just resolder the existing resistor would be my first guess.

Is something connected to GPIO15 | GPIO0 | GPIO2 ?

micropet, 2018-03-22:
I soldered the 10k.

All other pin are blank.

Now the Wemos can not be flashed anymore.

micropet, 2018-03-22:
I'm breaking up for today.
I'm cold. Here in the basement it is only 15 degrees Celsius

See you tomorrow and thank you.

micropet, 2018-03-22:
I found another 12 month old wemos.
He has never been in operation
Now I come to log the entire boot process:

FS   : Mount successful, used 0 bytes of 957314
RESET: Resetting factory defaults...
RESET: Warm boot, reset count: 0
RESET: formatting...
RESET: formatting done...
FILE : Saved config.dat
FILE : Saved security.dat
FILE : Saved config.dat
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

After pressing Reset-------------------------------

INIT : Booting version: (custom)
INIT : Warm boot #1
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:19824
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216-16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0

Exception (2):
epc1=0x3fff5ccc epc2=0x00000000 epc3=0x00000000 excvaddr=0x3fff5ccc depc=0x00000000

ctx: cont 
sp: 3fff4ee0 end: 3fff5360 offset: 01a0

>>>stack>>>
3fff5080:  00000000 4026e56f 3ffefaec 6c757201  
3fff5090:  3ffeb2b8 37363534 62613938 3fff5e42  
3fff50a0:  00000000 3fff53a4 000000e0 401004e4  
3fff50b0:  3ffeb2b8 3ffeb2b8 3fff5dfc 401075a4  
3fff50c0:  3fff01f0 4026ff68 402725cc 401043a2  
3fff50d0:  3fff5dfc 3fff5d2c 3fff5dfc 4010441e  
3fff50e0:  0000002e 3fff0320 3fff5110 40261cac  
3fff50f0:  00000000 0000ed47 40271f5f 3fff5801  
3fff5100:  3fff57cc 3fff57cc 3fff5dfc 40104508  
3fff5110:  40261ddf 3fff5dfc 3fff5e42 40261de8  
3fff5120:  00000008 3fff5150 3ffefb70 00000012  
3fff5130:  4026fef7 3ffefaec 00000012 3fff5e50  
3fff5140:  3fff5dfc 3fff5de8 3fff57cc 4026820d  
3fff5150:  005e0001 3ffefb00 00000002 00000000  
3fff5160:  00000018 3fff5de8 3fff5dfc 40268d70  
3fff5170:  40285662 3ffe85dd 3fff6934 00000000  
3fff5180:  3fff57d0 00000001 00000000 00000002  
3fff5190:  3fff572c 00000004 3fff5de4 402693f4  
3fff51a0:  40248571 4024854d 3fff696c 3fff5de4  
3fff51b0:  00000000 3fff5dfc 3fff5e68 40268db0  
3fff51c0:  3fff57cc 3fff51f0 00000004 3fff5de4  
3fff51d0:  00000016 3fff57cc 3fff5de4 40268541  
3fff51e0:  3fff57cc 3fff51f0 00000004 40261fa8  
3fff51f0:  00000494 00000000 3fff5924 40261fe4  
3fff5200:  3fff5de8 3fff57cc 3fff5d94 40268612  
3fff5210:  3fff5d74 3fff3bf0 00000010 00000020  
3fff5220:  3fff528c 3fff57cc 3fff5de4 402687eb  
3fff5230:  3fff528c 3ffeb3f4 3fff57cc 40268877  
3fff5240:  00000001 00000001 3fff52c0 00000001  
3fff5250:  3fff3d9c 3fff3bec 3fff3d9c 4024e822  
[... 189 lines cut ...]
INIT : Warm boot #4
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
CRC  : No program memory checksum found. Check output of crc2.py
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK 
INIT : Free RAM:19824
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP-216-16 with address 192.168.0.250
WIFI : Connecting SMC attempt #0
OTA  : Arduino OTA enabled on port 8266
EVENT: System#Boot
WD   : Uptime 0 ConnectFailures 0 FreeMem 16488
WIFI : WiFi connect attempt took: 5751 ms
EVENT: Time#Initialized
Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
EVENT: WiFi#ChangedAccesspoint
EVENT: WiFi#Connected
WIFI : Connected! IP: 192.168.0.229 (ESP-216-16) AP: SMC AP BSSID: C8:0E:14:05:BC:C6
EVENT: Clock#Time=Thu,21:31
EVENT: Clock#Time=Thu,21:32
WD   : Uptime 1 ConnectFailures 0 FreeMem 15088

whatsupskip, 2018-03-22:
Can someone please change the title as it is too general. This seems to be a very specific case and not the pre-compiled version that is used by most people here.

micropet, 2018-03-23:
It affects all mega versions> 19.03.2018, also the precompiled ones.

micropet, 2018-03-23:
It works on both critical wemos !!

What does that mean?
MQTT: Intentional reconnect

---------------------------------------
MQTT : Intentional reconnect
MQTT : Intentional reconnect
MQTT : Intentional reconnect
WIFI : WiFi connect attempt took: 4765 ms
Current Time Zone:  DST time start: 2018-03-25 02:00:00 offset: 120 minSTD time start: 2018-10-28 03:00:00 offset: 60 min
EVENT: Time#Initialized
EVENT: WiFi#ChangedAccesspoint
EVENT: WiFi#Connected
WIFI : Connected! IP: 192.168.0.216 (ESP-216-16) AP: SMC AP BSSID: C8:0E:14:05:BC:C6
EVENT: Clock#Time=Fri,17:59
MQTT : Intentional reconnect
MQTT : Connected to broker with client ID: ESPClient_68:C6:3A:A5:F1:9E
Subscribed to: ESP-216/#
EVENT: MQTT#Connected
EVENT: Clock#Time=Fri,18:00
WD   : Uptime 1 ConnectFailures 0 FreeMem 16440
done.
WD   : Uptime 1 ConnectFailures 0 FreeMem 16232
EVENT: Rules#Timer=1
ACT  : Publish ESP-216/IP,192.168.0.216
ACT  : Publish ESP-216/Time,18:00:31
ACT  : Publish ESP-216/Uptime,1
ACT  : Publish ESP-216/RSSI,-60
ACT  : timerSet,1,60
EVENT: Clock#Time=Fri,18:01
WD   : Uptime 2 ConnectFailures 0 FreeMem 16232
WD   : Uptime 2 ConnectFailures 0 FreeMem 16232
EVENT: Rules#Timer=1
ACT  : Publish ESP-216/IP,192.168.0.216
ACT  : Publish ESP-216/Time,18:01:32
ACT  : Publish ESP-216/Uptime,2
ACT  : Publish ESP-216/RSSI,-61
ACT  : timerSet,1,60

TD-er, 2018-03-23:
I got it reproducible and I guess it has something to do with the fact that having a connection doesn't mean you've got an IP address.

I will fix it :)

micropet, 2018-03-23:
Fine.
Thank you for your effort.
I hope I had not led you too much in the wrong direction. :)

TD-er, 2018-03-23:
But just curious, I assume you got todays patches loaded and those do work fine now?

At the moment I kinda dug into the WiFi thing quite deep and there is currently a almost rewrite of the WiFi, based on events instead of polling.

micropet, 2018-03-24:
Good Morning.

I compiled yesterday's source and it works.
I look forward to the event Rewrite.

[1 more reply]

closed 2018-10-13 by TD-er