letscontrolit/ESPEasy #1152: Text Scaling in OLED LCD SSD1306

whatsupskip, 2018-03-22:
This is an enhancement.

### Expected behavior
The text is really tiny and just about unreadable.

### Actual behavior
The display should be able to show one or two lines of text. If one line then takes up the whole vertical height of the screen. If two lines, then only half the screen.

### System configuration

Hardware: 0.91 inch OLED module 128x32 (i2c)
Wemos D1 mini, but would apply to all ESP8266

Software or git version: v2.0-20180227

TD-er, 2018-03-23:
Just to make sure, are you using OLED or OLED Framed plugin?

whatsupskip, 2018-03-23:
If I remember correctly the framed driver version didn't work at all.

What does "framed" actually mean in this case?

TD-er, 2018-03-23:
It adds a 'frame' around the information.
The top line (10 px) contains 

- time
- hostname/WiFi SSID (alternating)
- WiFi strength

The bottom contains pagination symbols (there are requests for redesign of this plugin)
And inbetween you can have 1,2, 3 or 4 lines of text. Nr. lines per page determines font size.
With up to 12 lines possible, you can have multiple pages.

whatsupskip, 2018-03-24:
Thanks that makes sense.
It was the unframed version then.

TD-er, 2019-11-04:
The Framed version (which is also being improved in #2706 ) does support larger fonts, so this one can be closed.

closed 2019-11-04 by TD-er