letscontrolit/ESPEasy #1153: Device Categories & Names In Device Drop Down List

whatsupskip, 2018-03-22:
This is an enhancement (in my opinion anyway).

As the list of supported devices is growing so well (great to see), some of the categories don't quite make sense.

Examples:
Dust & Gas: These are environment sensors and there are combined temperature, humidity and gas sensors like the BME680.

Suggestions:
As some category names are becoming quite long, some should be abbreviated, such as "Environment" becomes "Enviro" and "Communications" becomes "Comms". This will probably be very important for mobile screens where I suspect the key name may be off the screen on some smaller mobile devices.

In the environment category the sensors could then be listed with sub-categories such as "Dust", "Gas" and "Climate". 

Keep up the great work devs and contributors!

Grovkillen, 2018-03-23:
I think we will add a category list and after a first selection of category a second list is presented with devices for that perticular category. Something for the future though.

whatsupskip, 2018-03-23:
Ok, clearly you have thought about this already. Great!

closed 2023-09-04 by TD-er