letscontrolit/ESPEasy #1154: Sensor Power INA219 Labels for I2C Addresses

whatsupskip, 2018-03-22:
Enhancement

Hardware: Most common breakout board for [INA219](https://www.aliexpress.com/item/EYEWINK-Zero-drift-CJMCU-219-INA219-I2C-interface-Bi-directional-current-power-monitoring-sensor-module/32815859340.html?spm=a2g0s.9042311.0.0.JdlXCh) 

Software or git version: mega-20180316
I assume it applies to the other more recent releases.

I just thought it would be useful if the other related addresses were included with the i2c addresses:
Board Jumper A0 is 0x41
Board Jumper A1 is 0x45

TD-er, 2018-03-23:
In theory the usable I2C addresses for the INA219 are 0x40 - 0x4F
[image]

whatsupskip, 2018-03-23:
I know, but most people, I assume >80%, would be using the standard break out boards.

PR #3994 (2022-04-03): [P027] allow all possible I2C addresses
From a [forum request](https://www.letscontrolit.com/forum/viewtopic.php?t=9079)
Resolves: #1154

- Add all technically possible I2C addresses (0x40 .. 0x4F)
- Add list of required A0/A1 connections for addressing to documentation
- Uncrustify the source
closed 2022-04-03 by TD-er