letscontrolit/ESPEasy #1155: No control over MQTT Client ID in controller

okd475, 2018-03-23:
### Steps to reproduce
**How can we trigger this problem?**
Visible every time ESP connects to the MQTT broker (in MQTT broker logs)

**Does the problem persists after powering off and on? (just resetting isn't enough sometimes)**
Yes

### Expected behavior
Client ID of MQTT client for controller should use %sysname% as in case of MQTT client for Generic MQTT import device (or there should be possibility to define it manually - at least set some prefix).

### Actual behavior
It seems that the ESP uses separate MQTT clients - one for the controller (like Open HAB) and one for Generic MQTT Import device. As far as the client ID of the Import device is set to the "%sysname%-Import", client ID for controller is set to hardcoded "ESPClient_xxxxx" with no control over it (apart customized build).
Such behavior causes that simple MQTT broker security mechanism of accepting only the clients with predefined prefix (clientid_prefixes option in case of Mosquitto broker on RPi) fails - for %sysname% containing desired prefix, ESP receives MQTT data via Generic MQTT Import device correctly, but is not able to respond using configured controller (thus also via Publish function in rules) as it does not have required prefix in the client ID ("ESPClient_xxxxx").

### System configuration

Hardware:
ESP-01 (1024k)

Software or git version:
ESPEasy_v2.0-20180220

TD-er, 2018-03-23:
So to summarize:

- Format of client ID of controller is OK
- Format of client ID of MQTT importer is not

Should the client ID of the MQTT importer be the same as for MQTT controller, or fully customizable?

okd475, 2018-03-23:
Rather the other way around:
- Format of client ID of MQTT importer is OK as it is set basing on the %sysname% - so you can control it
- Format of client ID of controller is NOK - it is hardcoded and there is no control on it

I guess it would be enough to use %sysname% as the base for both client IDs as a simpler solution - for example "%sysname%-Import" and "%sysname%-Controller" (instead of current hardcoded "ESPClient_xxx").

Alternatively there could be added full customization of MQTT client IDs for both - controller(s) and importer(s).

TD-er, 2018-03-23:
As I understood, there should be some guarantee the client IDs are unique.
That's why I added the MAC.
I wil have a look at the current space used in the controller settings to see how much space is left.
There was also a request for longer username/password for some controllers if my memory serves me right.

okd475, 2018-03-23:
Yes - client IDs must be unique within single MQTT broker not to generate any unexpected system behavior, so the "Should the client ID of the MQTT importer be the same as for MQTT controller" is rather not an option.

I would say that replacing controller ID from current "ESPClient + MAC" (I guess it is what you meant) to something like "%sysname% + %unit%" should solve the issue keeping the needed space not to big.
What do you think?

Grovkillen, 2018-03-23:
It should be an option to alienate between
`ESPClient-MAC`
and
`Sysname-Unit`
With current MAC identifier being the default.

I think the physical unit is above in hierarchy compared to the user based.

TD-er, 2018-03-23:
Why not Sysname-MAC?
Or do you have to grant access per node?

Grovkillen, 2018-03-23:
ESPClient-MAC would let you grant access to all ESPClient* the Sysname-Unit would let you replace physical units and knowing in before hand a unit ID since you configure it yourself. Both ways have their ups and downs. Sysname-MAC could also work but that works be a third option.

TD-er, 2018-03-23:
I think I could find a spare byte to store these preferences. 
Do you think you guys can think of more than 255 possible format options for the client-ID?

Grovkillen, 2018-03-23:
I'm happy.. :wink:

TD-er, 2018-03-23:
Well then please start a descriptive list for the pull-down menu. :)

okd475, 2018-03-23:
Second part (MAC/Unit/whatever) is rather not so important I would say, so it could stay as it is right now.
Using Sysname in both client IDs would be probably the easiest/quickest/cleanest solution as you are already able to change it - no additional menus/check-boxes/etc are needed.

Grovkillen, 2018-03-23:
Well, both should have the same since they are published/subscribed from the same unit? Or am I misunderstanding something here?

The dropdown menu could look something like this:

```
Default
MAC
Sysname + MAC
Sysname + Unit
```

And what ever you choose the actual ID is then presented in the GUI...

denisfrench, 2018-03-23:
+1, so long as one of the options is the plain %sysname% (without the suffix). Yes, the user will need to deconflict (so don't make that the default), but there needs to be a way to set the Client ID directly.

See also #431 #436 and #925 . Should these issues be merged?

TD-er, 2021-08-12:
Is no longer an issue, as it has been fixed a long time ago.
Close this now.

closed 2021-08-12 by TD-er