letscontrolit/ESPEasy #1156: PWM flicker

DittelHome, 2018-03-23:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
PWM,pin,level results in a flicker on the connected LED. This is i known problem (i think), but i cant find a issue for it.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
yes

### Expected behavior
Tell us what should happen?
PWM without flicker

### Actual behavior
Tell us what happens instead?
PWM with heartbeat problem, flickering.

### System configuration
LCD display LED is connected to GPIO 5.
Flickerfree values are only 2, and 200 above.
Values from 3 until 200 are unusable. From 200 and above the LED is very bright.

Hardware:
wemos d1 mini

Software or git version:
ESPEasy_mega-20180323

TD-er, 2018-03-23:
Just to summarize:

- Flickerfree values are only 2, and 200 above.
- When set, the output is constant, no periodical flickering? Or is it hard to detect at the outer ranges?

DittelHome, 2018-03-23:
> Flickerfree values are only 2, and 200 above.

Absolutely correct

> When set, the output is constant, no periodical flickering? Or is it hard to detect at the outer ranges?

pwm,5,2 ==> Result OK (no flickering)
pwm,5,2 to pwm,5,200 ==> Result **not** OK (flicker)
pwm,5,200 to pwm,5,1023 ==> Result OK (no flickering)
In other words, flickering appears only in range form 3 to round about 200. I think the flickering is unremarkable at higher pwm values, since the LED is already very bright.

DittelHome, 2018-04-28:
Problem ist fixed with Core 2.4.x  !!!

closed 2018-04-28 by DittelHome

Grovkillen, 2018-04-28:
Aaaah! :heart_eyes:

DittelHome, 2018-09-25:
Thats correct, problem was only fixed with core 2.4.1.  I hope that the developer goes back to 2.4.1.

Feuerreiter, 2018-09-25:
Thanks for the answer. I try build with 2.4.1, and pwm works fine.