letscontrolit/ESPEasy #1157: ESPeasy go to Thingsboard

gdrako, 2018-03-23:
There is a possibility that the Thingsboard platform will be included in the list of controllers in the future??
For more information, look here ..https://thingsboard.io/docs/api/

Regards

gdrako, 2018-04-17:
Hello
There is a timeline for this implementation ??
Thank you

TD-er, 2018-04-17:
Not yet, we're currently working very hard towards the new release.

doleron, 2018-06-14:
There is any workaround to use espeasy with thingsboard?

WORX-Solutions, 2018-08-08:
A thingsboard implementation will be great!

datenheim, 2019-10-14:
Hm, played around with thingsboard a while... and now touched ESPeasy :)
Would really like to see it integrated (if it not already is).

In fact it should be easy to integrate - the only trouble is thingsboard gave
all the mqtt-things different names :)

When you create a device in Thingsboard it gives you a "device id" and "access token".
The **device-id must be used as mqtt client id** and
the **access token must be used as mqtt user name** while the **mqtt password must stay empty**.

Also the topics for sending data (telemetry) and properties (attributes) of the device are pre-defined, also writing something back from Thingsboard to the device goes via pre-defined RPC topics which the device must subscribe to and confirm actions to another rpc-topic.

As a first test the device should send its measurements to topic
_v1/devices/me/telemetry_
as json string, either as plains (data gets servers timestamp of transmission):
_{"value1":1.2,"value2":23}_
or with clients time stamp as:
_{ts:#unixtime#,values{"value1":1.2,"value2":23}}_
and thus allows sending historical data etc.

State information (data that must not be stored in the database forever) can go to
_v1/devices/me/attributes_
as a plain json string.

You can also check that with mqttfx or such. If you need thingsboard access to a device for testing I may help with that.

datenheim, 2019-10-16:
> I was just glancing at the [MQTT format](https://thingsboard.io/docs/reference/mqtt-api/) of this Thingsboard.
> It looks like you have to supply some key name and a value.
> My initial guess is that you could add a standard topic for all sensors on a node (set at controller level in ESPeasy) and use the value name as key.
> But then you have to make sure no duplicate names are used in the plugins. Or else you should make a combination of plugin-name + value name.
> Do you have any idea on limitations on Thingsboard when using this approach?
> For example key length maximum, of forbidden characters?"

Yes, that should work. The topic for all sensor values is fixed, it is simply "v1/devices/me/telemetry" and each value should be sent as a key in the values part section:
{"temperature1":12.3,"temperature2":23.6,"humidity":85.2}
For sure fancy characters may pose problems and I would recommend to limit yourself to names that also work well in typical programmnig languages.
 
> Is there some upload limit in messages per time unit like with Thingspeak?

No. As you can install the free thingsboard version on any system you like only network and system power define the limit. They promise the microservice-installation is fit for 100.000+ devices :)

> Should the timestamp be included by ESPeasy? (only when NTP was successful I guess)

This depends, you have the choice. If you do not want to provide a timestamp the data gets a timestamp from the server on reception. in this case you only provide a JSON string as noted above.

But if you have the real time available on the sensor, you may send it with the data, the JSON string then looks like this:
{"ts":16000000000,"values":{"temperature1":12.3,"temperature2":23.6,"humidity":85.2}}

This would allow to cache a number of readings on the sensor, only connect from time to time with the WiFi/GSM/Lora to save power and then send all cached readings each with it's own timestamp. This is a scenario I am using.

Please give a hint how/where to start to implement a new controller protocol!
I really like to get that soon ;-)

chunter1, 2023-04-30:
Since the topic is rather old i wonder if it is already possible to connect ESPEasy with Thingsboard?

tonhuisman, 2023-04-30:
> Since the topic is rather old i wonder if it is already possible to connect ESPEasy with Thingsboard?

There is no specific Controller for Thingsboard included with ESPEasy, but using the Home Assistant MQTT controller (that is a quite generic and versatile MQTT controller), and applying the instructions above, should get you in the right direction.

tonhuisman, 2023-04-30:
And if the MQTT route fails, you can use a recent build of ESPEasy for the PostToHTTP command, and use the [HTTP API of Thingsboard](https://thingsboard.io/docs/reference/http-api/) via rules.

chunter1, 2023-04-30:
Ok, i got the data upload part using MQTT and a publish rule working.
However, should the direction from server to client also work?
I created a shared attribute in thingsboard and entered "v1/devices/me/attributes" in controller subscribe config field.
When looking at the log, no data is coming in when changing the attributes value in thingsboard.

tonhuisman, 2023-05-01:
> However, should the direction from server to client also work?

Controllers in ESPEasy are used to send out data to external systems.
To get data in you will need a plugin (Devices), so you will need the [MQTT Import plugin](https://espeasy.readthedocs.io/en/latest/Plugin/P037.html#p037-page) for that. Depending on the build you have installed, some options may not be available.

TD-er, 2023-05-01:
C005 HomeAssistant/OpenHAB MQTT does support receiving some commands.

See the [documentation](https://espeasy.readthedocs.io/en/latest/Controller/C005.html#c005-page) for some examples.

It is by far the easiest to trigger events with some event values which are then dealt with in rules.

chunter1, 2023-05-01:
Thank you for your support!
I'll take a look at C005 HomeAssistant/OpenHAB MQTT.

datenheim, 2023-06-13:
With thingsboard 3.5 and maybe already 3.0 any generic MQTT publishing could be connected to thingsboard because it now allows to modify the topics, also allows to connect with plain MQTT auth, ... and the rest can be done in the rule chains.
I successfully connected tasmota and so I'm sure ESPeasy too will work with thingsboard now.
Back than it seemed impossible because TB was missing the flexibility....
I think this could be closed now (again).

closed 2023-06-13 by TD-er