letscontrolit/ESPEasy #1162: Possible to add ESP32 build to the nightly auto builds?

Budman1758, 2018-03-23:
I notice that there are fixes and stuff for the ESP32 once in a while in the commit notes.  Is it possible to include ESP32 in the nightly builds? Or at least when changes or commits get merged that are relevant? I know there is some debate about the future of ESP32 vs UpyEasy but for now it would be nice to test the changes and fixes.
I would like see the development of ESP32 to continue.  While I have nothing against UpyEasy I fear it won't be quite as "easy" as ESPEasy.

Grovkillen, 2018-03-23:
I leave that to @mvdbro to decide if a dev version is ready for the public.

redskinhu, 2018-03-23:
+ 1 vote  :-)

Budman1758, 2018-03-23:
@Grovkillen The version available on the ESP32 page is "public".     Just sayin.....  :-)
https://www.letscontrolit.com/wiki/index.php/ESPEasy32#Firmware

micropet, 2018-03-24:
1+

Grovkillen, 2018-03-24:
Sure, I don't think @mvdbro got nothing against a nightly build. :)
But it's his area :+1:

redskinhu, 2018-03-24:
Hi

Or someone could share platformio.ini which contains the setting to compile ESP32 version of ESPEasy? Or tutorial if it not so simple?

The compiled version little bit buggy. Some of bugs are fixed in the recent sources and we can compile custom fw too.

Thanks

flexiti, 2018-03-24:
1+

everuss, 2018-03-25:
+1
I would also like to see the platform.ini also
I would like to try if possible I can get other plugins from playground working. In particular the nextion plug in. Thanks

mvdbro, 2018-03-25:
You'd have to ask Edwin. My platform.io/travis build skills are zero. Can't be of any help.

Grovkillen, 2018-03-25:
That's all right, but is it ready for nightly?

mvdbro, 2018-03-25:
It would also require other devs to adopt ESP32 in their environment, otherwise you will likely be confronted with daily build failures.

TD-er, 2018-03-25:
I have a few ESP32's at stock and room at my sophisticated test lab:
[image]

Grovkillen, 2018-03-25:
Is that sun block I see? :)

TD-er, 2018-03-25:
Yep, for when I have a bright moment ;)

Grovkillen, 2018-03-25:
It must be empty by now then?

Budman1758, 2018-03-25:
Just curious....  Why not an ESP32 repository?  Seems like that would help keep things organised a bit better.

TD-er, 2018-03-25:
That would make development merges between ESP8266 and ESP32 almost impossible.

And they differ not that much when developing new controllers and/or plugins.

Zixim, 2018-04-06:
+1 for regular builds for esp32.

TD-er, 2018-08-07:
It is already implemented, so closing this issue

closed 2018-08-07 by TD-er