letscontrolit/ESPEasy #1166: Is it possible to use text in the MQTT data fields?

rlust, 2018-03-24:
Would be great it some text could be passed to the MQTT import data fields.

Thanks for the consideration!
Great work here!

Software or git version:

Grovkillen, 2018-03-24:
We have no intention in letting device/task values become anything but float values. So no MQTT import will not get that feature. BUT that being said, the idea of importing strings from MQTT is something we're considering but the implementation will be in the controller side of the code. That's for the future though. :sunglasses:

Oxyandy, 2018-03-24:
Not sure if this helps, I use the publish command to send data from my ESP8266 module to be logged in Domoticz, I output that as TEXT and it is great way for me to see at glance important data I need..
Bonus of doing it this way as a TEXT log in Domoticz has a 'Clear' button, some of the graphs in Domoticz lack this option & so far I have no easy way to reset the data
Example:
```
on button#switch do
if [button#switch]=1
gpio,12,1
publish domoticz/in,{"idx":26,"nvalue":0,"svalue":"Pump_ON_%systime%"}
else
gpio,12,0
publish domoticz/in,{"idx":26,"nvalue":0,"svalue":"Pump_OFF_%systime%"}
endif
endon
```
_And haha thanks, you just solved a tiny problem I had, 
I guess in the process of copying, pasting the rule around, I cut off the "endon", 
oops off to fix that now.._

Oxyandy, 2018-03-24:
Yeah, yeah 'Import', I over looked & also wasn't 100% sure on usage, so I was just again reading up..
https://www.letscontrolit.com/wiki/index.php/MQTT_Import
So if you imported a number, couldn't that be converted to text in a rule ?
Again not sure if helps, just throwing around ideas ;)

tonhuisman, 2023-09-03:
MQTT Import plugin has support for mapping strings to numeric values, so this can be said to be solved, and can be closed.

closed 2023-09-04 by TD-er