letscontrolit/ESPEasy #1168: Update for smaller displays

sentinel073, 2018-03-24:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
How can we trigger this problem?
Resize the window: when reducing the width, at some point the menu items will show one below the other

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
--> yes

### Expected behavior
It is an enhancement: make the menu items appear in 2 rows and also adjust the threshold (to 780). Replace the code in WebServer.ino as follows:
`    "@media screen and (max-width: 780px) {"`
`      ".bodymenu{  margin-top: 11px; margin: 0px;}"`
`      ".headermenu{  position: relative;   height: auto;   float: left;   width: 100%;   padding: 7px 10px; }"`
`      ".headermenu h1{  padding: 8px 12px; }"`
`      ".menubar{  top: 0px;   position: relative;   float: left;   width: 100%; }"`
`    "\0"`

### Actual behavior

### System configuration

Hardware: wemos d1 mini

Software or git version: ESPEasy-mega-20180319

whatsupskip, 2018-03-29:
Please change the title to make it clearer what is meant, such as "Update web css for smaller displays". 
I thought the topic was about OLED displays connected to ESP8266

sentinel073, 2018-03-29:
done :-)

tonhuisman, 2023-09-02:
This has been taken care of, so can be closed.

closed 2023-09-02 by TD-er