letscontrolit/ESPEasy #1170: Encoder plugin bug (with encwrite command)

giig1967g, 2018-03-25:
The new encwrite command (issue #1086), the item value number is not updated in the device page, so it shows incorrect values in OLED or LCD.

### Steps to reproduce
Create an encoder device.
Use the new ENCWRITE command.
The value updated with ENCWRITE is not updated in the device page.

### Expected behavior
The value updated with ENCWRITE should update the value in the device page.

### Actual behavior
The value updated with ENCWRITE is not updated in the device page.

### System configuration
Wemos D1
mega-20180324

Hardware:

Software or git version:

uzi18, 2018-03-25:
fixed in PR #1150

giig1967g, 2018-04-29:
The solution proposed by @uzi18 doesn't work.
The reason is that inside PLUGIN_WRITE the variable [event->BaseVarIndex] is not set and is always = 0.
So actually this PR modifies the 1st value of task 1 and not the encoder counter.

My workaround so far is to add a taskvalueset just after the encwrite command.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-04 by TD-er