letscontrolit/ESPEasy #1171: System variables are not processed when executing commands through web

Grovkillen, 2018-03-25:
If I run a publish command for example the %systemvariablename% is not processed. See pictures

[image]
[image]

Grovkillen, 2019-10-10:
Is this still an issue?

TD-er, 2019-10-10:
I would be really surprised if it is still an issue.
The commands should all be executed by the same rules engine now.
You could test it using the logitem command :)

TD-er, 2019-10-27:
Just tested with `logentry,%mac%` and it does show the MAC-address in the log.
So it is working and thus this issue can be closed.

closed 2019-10-27 by TD-er