letscontrolit/ESPEasy #1174: Degree sign on LCD1602/2004 via MQTT and HTTP

elmicha, 2018-03-25:
### Steps to reproduce

Thanks to https://github.com/letscontrolit/ESPEasy/pull/983 the degree sign now works if it is set from the Web GUI. 

    curl "http://esp_easy-3/control?cmd=lcd,4,1,27%DFC"
    echo "lcd,4,1,27\xDFC" | mosquitto_pub -t ESP_Easy/3/cmd -l

also works from HTTP and MQTT, but it would be nicer if we could use a UTF-8 degree sign.

Currently, `curl "http://esp_easy-3/control?cmd=lcd,4,1,27°C"` shows some Chinese characters.

### System configuration

git version mega-20180324

TD-er, 2018-03-25:
That simply means the UTF-8 character doesn't get to the webserver as you expect it would.
The UTF-8 version of the character is supported. You can try to enter it in the text field of the plugin.
I am not sure how an Unicode character gets encoded via a GET request. It may even be browser dependent?

elmicha, 2018-03-25:
Are you sure that it works with these 20x4 displays? I'm not talking about a problem with HTTP or MQTT generally, but specifically with these displays. They use a special charset, and for the degree sign you have to use 0xDF (only one byte, no UTF-8).

[This change](https://github.com/letscontrolit/ESPEasy/pull/983/commits/7a4e3d954d6cc8525215d33b9693186c7b12ad6a) made it work from the web GUI.

TD-er, 2018-03-25:
I don't have such a display.
But I wrote the detection/translation part for all special characters on the oled display and all what's being done for this LCD plugin is to hook into the already present character replacement functions to replace it with the byte(s) referencing that symbol in the display font.

So I am fairly sure the route up to the display plugin is covered and the photo with the pull request suggests it should work.
And if it works using {D} it will also work with the utf-8 encoding for the degree symbol, since {D} is internal replaced into the utf-8 sequence and then processed further.

elmicha, 2018-03-26:
I plugged in a 128x32 OLED and can confirm that it works like you wrote:

        mosquitto_pub -t ESP_Easy/3/cmd -m "OLED,4,1,from MQTT: 27°C"
        mosquitto_pub -t ESP_Easy/3/cmd -m "OLED,4,1,from MQTT: 27{D}C"
    curl "http://esp_easy-3/control?cmd=OLED,4,1,from+HTTP:+27°C"

all show the degree sign. But for the LCD:

    mosquitto_pub -t ESP_Easy/3/cmd -m "lcd,4,1,from MQTT: °C"

shows Chinese characters instead of the degree character.
    
    mosquitto_pub -t ESP_Easy/3/cmd -m "lcd,4,1,from MQTT: {D}C"

shows literal "{D}" instead of the degree character.

Your working P023_parseTemplate() is exactly the same as the P012_parseTemplate(), so the problem must be somewhere else. Oh: P012_parseTemplate() is only called once in the "case PLUGIN_READ", but you are calling it also in "case PLUGIN_WRITE".

With this change it works:

    diff --git a/src/_P012_LCD.ino b/src/_P012_LCD.ino
    index d29a858..ba586e4 100644
    --- a/src/_P012_LCD.ino
    +++ b/src/_P012_LCD.ino
    @@ -244,8 +244,9 @@ boolean Plugin_012(byte function, struct EventStruct *event, String& string)
             else if (lcd && tmpString.equalsIgnoreCase(F("LCD")))
             {
               success = true;
    -          argIndex = string.lastIndexOf(',');
    -          tmpString = string.substring(argIndex + 1);
    +          tmpString = P012_parseTemplate(string, cols);
    +          argIndex = tmpString.lastIndexOf(',');
    +          tmpString = tmpString.substring(argIndex + 1);
     
               int colPos = event->Par2 - 1;
               int rowPos = event->Par1 - 1;

I'm sorry, I don't know yet how to make a pull request. Thank you for your help!

TD-er, 2018-03-26:
Should be in there in the next build :)

TD-er, 2018-10-13:
I'll close this one now.
If it is still a bug, please re-open.

closed 2018-10-13 by TD-er