letscontrolit/ESPEasy #1176: Is is possibe to increase the MQTT topic lengh in the MQTT Inport plugin?

redskinhu, 2018-03-26:
**NOTE: This is not a support forum! For questions and support go here: https://www.letscontrolit.com/forum/viewforum.php?f=1**

### Steps to reproduce
The MQTT Topic input field lenght is only 40 character lenght.

Does the problem persists after powering off and on? (just resetting isn't enough sometimes)
N/A

### Expected behavior
Could it be 64 character lenght (or more)?

### Actual behavior
Tell us what happens instead?
The MQTT Topic input field lenght is 40 character. And some MQTT topic couldn't fit in the field.

### System configuration

Hardware:
All ESP
Software or git version:
All

tonhuisman, 2022-04-17:
In PR #3424 a global topic prefix of 40 characters has been introduced. Despite that this requires that all used topics use the same prefix, it could be a solution for this issue.

tonhuisman, 2023-09-01:
No response for a long time, so probably can be closed...

closed 2023-09-01 by TD-er