letscontrolit/ESPEasy #1177: Add the ability to install plugins in runtime - ENHANCEMENT

giig1967g, 2018-03-26:
Hi, 
this is an enhancement request.

I believe that a plug-in should be really a plug-in, meaning a piece of code that can be be added to the core in runtime.
The expected behaviour should be the possibility to load in the unit only the plugin that one really uses. 

Benefits:
1) it saves memory 
2) there is no need to recompile the whole firmware if a bug in a plugin is fixed 
3) playground plugins could be tested without having to manually compile the firmware 
4) the core will remain separated from the plugins

What do you think?

TD-er, 2018-03-26:
I think that would be great, but near to impossible given the current way C/C++ is being compiled into a firmware for the ESP8266.
The only way it could be done, is when the plugin was written in some interpreter language for which the ESP8266 should run the interpreter.
That would take too much development time and we would loose a lot of flexibility and hardly gain anything.

Other programming languages do offer support for runtime loading of code and also for C/C++ exist techniques for this, but it is not possible in the current setup.
Maybe uPyEasy (based on Python, in early beta stage) can do something like this.

uzi18, 2018-03-26:
It is possible, but too much effort needed.
Entire build system must be changed, and now core is realy modded if You look in comparison to legacy version.

giig1967g, 2018-03-27:
I was thinking of a sort of .dll like structure.
Where every plugin has it's own function calls.
I would avoid internal interpreter as that would slow down the execution...

Anyway, was just an idea.

BTW, you are doing an excellent job!!

M0ebiu5, 2018-03-30:
> it saves memory

 it would save flash memory (where we have enough on 4MB devices), but increase ram usage (which is already high)
> there is no need to recompile the whole firmware if a bug in a plugin is fixed

 and what about the handling of the plugins (up-download, versioning, checks, etc)
> playground plugins could be tested without having to manually compile the firmware

 you would still need a mechanism to exclude unstable plugins from production

- Why not focus on the frequently used and demanded features?
- Or increase stability (some of my devices crash after apr 75000 min)?
- Let plugins register their own timers, instead of the hardcoded entry points?
- Make some thoughts about automated (stress-)testing to enhance espeasy to a higher level?

TD-er, 2018-03-30:
@M0ebiu5 They probably crash after approximately 71582 minutes, give or take a few :)
And which version do you use, since that bug was fixed in december and a few of my devices had uptimes over 55 days since then. (all devices that were not affected by a power-dip)

A registering their own timers, that's exactly what I am changing the last few months.
Starting with MQTT and WiFi, but also other things are now running asynchronous.

@psy0rz has built a test-rig to run unit tests each night, but it is not operational yet. When it is, it will be shown.

So I can honestly say all these bullets can be marked with either a check or a "working on it". :)

M0ebiu5, 2018-03-30:
@TD-er i'm pretty sure they have the bug you describe. But cause this is the only and easy solvable problem, i can live with it :)

>So I can honestly say all these bullets can be marked with either a check or a "working on it". :)

Thats great!

tonhuisman, 2022-05-07:
This is not likely to be implemented soon, because of lack of support from C++, so can best be closed. Let's keep it on a back-burner though, you can never know what could become available 😄

closed 2022-05-07 by TD-er