letscontrolit/ESPEasy #1181: Feature Request CPLUGIN_PROTOCOL_ONCE_A_SECOND

s0170071, 2018-03-27:
The thingspeak controller as it is today does not obey the 15 seoncs rule that says that you must send only one message every 15 seconds. If you send in shorter intervals,  the message will be ignored, the data is lost.

I want to refurbish the thingspeak protocol so that it better handles 
- the maximum message rate of 1 per 15 seconds
- allocation of  simultaneous readings from multiple sensors into one message.

For that  I need the protocol plugin to be re-scheduled at certain intervals, e.g. every second. This way I can buffer thingspeak requests and allocate data.

TD-er, 2018-03-27:
What are you planning to do when you receive a sample from the same plugin within the 15m interval?
Only send the last one?
Can you combine values into a single 'push'?
How and where should this be defined what value is being sent?

s0170071, 2018-03-27:
> What are you planning to do when you receive a sample from the same plugin within the 15m interval?

Its 15s and it would be sent as soon as the 15s elapsed. 

> Only send the last one?

Yes. If you send more often, no value would get through.

> Can you combine values into a single 'push'?

Yes. Each ThingSpeak channel has 8 fields that can be sent in a single request. 

> How and where should this be defined what value is being sent?

Just like it is today. The idx value of a sensor defines the offset of the thingspeak fields. 
Example: Controller 1 is thingspeak, you have a temp/hum sensor with two values and a dummy sensor with the uptime. 
Then you select idx= 1 for the temp/hum sensor and idx=3 for the uptime dummy. In Thingspeak those values end up in fields 1,2,3 as temp, hum, uptime. 

That the controller does when it is finished is: 
- it gains a 8- value buffer (type: double) 
- after a value update from a sensor it delays the thingspeak push request for 1 (more) second. If whitin that second further values are updated they are added to the next request (see example above)
- The request is held back a little more if the 15 seconds from the last request have not elapsed yet. During that period you can further update values to the controller. 
- optional: force send if all values(fields) have been updated once and 15s have elapsed. Otherwise you could delay the transmission forever if you update a value every second as this delays the transmission for a second...

TD-er, 2018-03-27:
Sounds like a well-thought solution ;)
Is this limit of 15 seconds for all users, or are there paid subscriptions which allow for more frequent updates?

Grovkillen, 2018-03-27:
https://thingspeak.com/prices/thingspeak_academic

Every second for paid account.

TD-er, 2018-03-27:
Maybe also add a selector to switch between all supported submit intervals?

uzi18, 2018-03-27:
https://thingspeak.com/prices/thingspeak_home
but still expensive ...

TD-er, 2018-03-27:
Indeed, looking at those prices, there must be quite a lucrative businessmodel.
It must be the compute time and the Matlab suite, because storing data isn't that expensive.
At my work, we're talking about the historic archive of data from ~60'000 sensor locations with aggregated data per minute of up to 60 measurement values per location.
And I am quite sure that's a bit less expensive, including hardware write-off, man hours, software to query that data, etc.

I should write a controller to interface with the software we sell and then we can get rich :)

Grovkillen, 2018-03-27:
Count me in :)

s0170071, 2018-03-27:
Well, u can't beat the free version, can you? I can pimp the controller but i would rather not touch the scheduler. Nor the configuration. TD-er, would you be so kind and prepare this ? I need that once a second state and the 15 second timeout variable if like that one configurable.

TD-er, 2018-03-27:
And need it in 15 seconds? ;)

s0170071, 2018-03-27:
No.... Just take your time and get rich first...