letscontrolit/ESPEasy #1183: LOGENTRY rules command - enhancement

giig1967g, 2018-03-27:
Hi,
I would like to suggest to add a new rules command to add a log entry in the system LOG in order to simplify the debug process.

The log command could be:
>> logentry,text

the "text" part should be able to evaluate the system variables like %systime% and the device values [task#value].

i.e. 
>> logentry,timestamp=%systime%
>> logentry,Device Value=[device#value]
>> logentry,free text

TD-er, 2018-03-27:
That's a nice feature indeed :)

giig1967g, 2018-03-27:
I am full of ideas.. just not able to code them properly :(
Anyway, thanks for your excellent work!

Grovkillen, 2018-03-27:
Ah, nice.

The output should then be something like this

Rule
```
logentry,timestamp=%systime%
logentry,Device Value=[device#value]
logentry,free text
```

Log 
```
215330404 : RULE: timestamp=22:19:08
215335476 : RULE: Device Value=24.2
215345965 : RULE: free text
```

TD-er, 2018-03-27:
Why log the timestamp, when the timestamp can also be metadata of any log line?

Oxyandy, 2018-03-27:
How would it differ from 'publish' which I use to debug rules already ?
```
on button#switch do
if [button#switch]=1
gpio,12,1
publish domoticz/in,{"idx":26,"nvalue":0,"svalue":"Pump_Start_%systime%"}
else
gpio,12,0
publish domoticz/in,{"idx":26,"nvalue":0,"svalue":"Pump_Stop_%systime%"}
endif
```

TD-er, 2018-03-27:
You can let logs write to a file using a syslog server

giig1967g, 2018-03-27:
exactly. not everyone use MQTT

Grovkillen, 2018-03-27:
The command should be `log_entry` I suggest. And instead of RULE maybe `RULE_LOG`?

giig1967g, 2018-03-27:
You have more experience than me, but I would maintain the same syntax of the current commands. 
So no underscore in commands, just Capital letters between the words:
LogEntry,text

Grovkillen, 2018-03-27:
What I mean with log2mqtt is a generic log not a manual log. I don't know if that's what @Oxyandy is referring to?

TD-er, 2018-03-27:
Why not simply `Comment,bla` ?
After all, it is just a comment in the output with the option to evaluate system variables.

Grovkillen, 2018-03-27:
Agree on the `LogEntry`. Thanks for keeping me in track 😎

Grovkillen, 2018-03-27:
Comment sounds a bit vague.

Oxyandy, 2018-03-27:
Comment is literal 
"idx":26 in Domoticz is a TEXT log stream, I dump everything there, 
which matters, I then see patterns of use and is easy to view everything in a glance

giig1967g, 2018-03-27:
I think that COMMENT does not reflect the fact that the text will go into the log

Oxyandy, 2018-03-27:
A LOg can be everything, a comment is just a snippet :P

Oxyandy, 2018-03-28:
Any prefix will do anyway, if it cuts the length of what I currently use down to just
What_ever, Pump_ON
I can then filter 'What_ever' in the syslog server output & view everything at a glance just as I do now, only with less effort & characters taking up valuable space in rules, yeah I like it.

giig1967g, 2018-05-05:
created PR#1374

Grovkillen, 2019-10-10:
This can be closed

closed 2019-10-10 by Grovkillen