letscontrolit/ESPEasy #1185: Feature Request: Task enable/disable via http commands

jobst, 2018-03-28:
It would be good to be able to turn off/on devices via a http command (Server -> Sonoff).

Say I have a number of (in my case) sonoff devices that control lights, and have on GPIO14 a PIR connected that drops the pulled up line to zero, which in turn is attached to Task 3, firing off a rule.

The motion sensor is not required during daylight hours that the server on the network "knows" through time zones. The server than could turn off the task in the morning, to turn it on again in the evening.

Suggested Command:

TaskEnable,<task number>,<1 or 0>

Alternatively:

TaskEnable,<task name>,<1 or 0>

IMHO the second one is prefered method.

Grovkillen, 2018-03-28:
Task number is the generic way and must be part of this command. The second suggestion using name is then built on top of that.

TD-er, 2018-03-28:
Why not introduce a check for sunrise and sunset?
Then it can be used in rules.
I guess 16 bit for longitude and for latitude is resolution enough to compute it to almost seconds.

TD-er, 2018-03-28:
There is already a request to disable plugins, controller etc. Remotely, but that was meant for troubling code and/or settings which prevents the user to operate or change settings

Grovkillen, 2018-03-28:
I use a simple photo resistor to see the outside light level. But that doesn't mean that a theoretical sunrise/sunset is not needed.

ostanki, 2018-04-01:
Very much want this feature !

Another usage example:
 On Clock#Time=All,18:25 do // every day at 18:25 start HC-SR04 device to monitor water level
  enable,1
 endon
On Clock#Time=All,18:45 do
 disable,1
endon

Because the HC-SR04 device might generate lots of traffic while refilling reservoir....

TD-er, 2022-04-21:
Is already fixed

closed 2022-04-21 by TD-er