letscontrolit/ESPEasy #1187: Add timestamp to serial log

giig1967g, 2018-03-28:
Hi,
I started using the serial log (connecting using Putty to the serial port) but I noted that there is no timestamp. I believe that the timestamp could be useful to investigate issues.

See example:
```
INIT : Booting version: mega-20180326
INIT : Warm boot #9
FS   : Mounting...
FS   : Mount successful, used 76806 bytes of 113201
CRC  : program checksum       ...OK
CRC  : Settings CRC           ...OK
CRC  : SecuritySettings CRC   ...OK
INIT : Free RAM:17240
INIT : I2C
INIT : SPI not enabled
EVENT: System#Wake
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESPT6-16 with address 192.168.4.1
IP   : Static IP :3456215232
WIFI : Connecting KeepOut attempt #0
EVENT: System#Boot
ACT  : gpio,14,1
SW   : GPIO 14 Set to 1
ACT  : gpio,12,1
SW   : GPIO 12 Set to 1
ACT  : gpio,13,1
SW   : GPIO 13 Set to 1
ACT  :
ACT  : taskvalueset,1,1,1
```

Grovkillen, 2018-03-28:
Many syslog servers add their own timestamps since the clients timestamps will not be in sync.

Oxyandy, 2018-03-28:
On windows I use Visual SYSLOG server, I tried plenty - this has filtering & timestamps 
[image]
If you are going to log multiple devices, you need better than putty offers

giig1967g, 2018-03-28:
You are right.
Instead of timestamp there could be the milliseconds since boot (like in the web log)

Grovkillen, 2018-03-28:
That's a better suggestion. Maybe you can edit the topic?

giig1967g, 2018-03-28:
done :)

TD-er, 2018-03-28:
The entire log needs a make-over.
It should be possible to add a topic and apply filtering on what to show.
Now it is just a filter on how much is sent to log, not on what is sent.
And also a lot of log texts need a lot of work to construct, but when the loglevel is not used on any log output, the log string is not being used. So try to avoid creating log when nobody is using it.

giig1967g, 2018-05-05:
done. can be closed.

closed 2018-05-05 by giig1967g