letscontrolit/ESPEasy #1188: Current Mega version can not be compiled under Arduino

micropet, 2018-03-28:
It is the unchanged Mega version of today. Arduino Ide 1.8.5
The following error messages are displayed:

D:\arduino-1.8.3\meine\ESPEasy-mega-03-28 ori\ESPEasy\ESPEasy.ino:77:32: fatal error: define_plugin_sets.h: No such file or directory
 #include <define_plugin_sets.h>
compilation terminated.

exit status 1
Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.

TD-er, 2018-03-28:
Those changes in include path were to test the better "scan for dependencies" option of PlatformIO.
But I guess they can be reverted since those scans did not give significant improvement on binary size.

And maybe we should have a discussion on the src directory.
It keeps causing issues with Arduino IDE.

micropet, 2018-03-28:
OK

susisstrolch, 2018-03-28:
What‘s about a simple symbolic link?

micropet, 2018-03-28:
Maybe you should note down such peculiarities somewhere.
Then the user knows and does not need to waste his time.

TD-er, 2018-03-28:
@micropet It wasn't intended to leave people with code that doesn't compile on a platform that I'm not using myself.

micropet, 2018-03-28:
It should not be a reproach.
But maybe you can test both platforms in the future.

TD-er, 2018-03-28:
I tried to just now, but on my laptop it took about 30 minutes to start the Arduino IDE.
I really have no idea why, because it is an i7 with 500 GB SSD and 8 GB RAM.

Maybe I should try some Linux version on another computer.

micropet, 2018-03-28:
<define_plugin_sets.h> is still wrong
it works with "define_plugin_sets.h".

micropet, 2018-03-28:
Oh, with me the IDE starts within a few seconds on all W10 computers.
On the server 2016, the start takes about 30 seconds.

micropet, 2018-03-28:
Mostly I work on Hyper-VM machines running on a Windows 2016 server.
They are very fast.

Budman1758, 2018-03-28:
I believe the official supported way to compile is with PlatformIO. Perhaps you would have less trouble with it?

micropet, 2018-03-29:
I find PlatformIO very confusing.
OVA updates often do not work that way.
Arduino is much cleaner and faster.

I just measured:
Arduino 1.8.5 takes 11 seconds to start and load all ESPEasy files.
Compile time is 25 seconds.

The computer is a 4 year old I7 without SSD under W10

Budman1758, 2018-03-29:
That's all well and good but methinks you missed the point of my post.

TD-er, 2018-03-29:
Good morning.
After -finally- an good night's sleep, I think I know what's causing the enormous loading times of Arduino IDE.
When creating a junction to the src directory, I made an error and created one to several levels below.
Since I didn't know how to remove it (I'm more a Linux guy) I renamed it and there is still a loop when you do recursive search.

Will try again this evening.

micropet, 2018-03-29:
Good morning.

I have the source file with me at / Arduino / ESPEasy-2018-03-28 / ESPEasy
and the libraries under / Arduino / libraries
Of course I have to copy them from the .zip File there before compiling.

micropet, 2018-03-29:
@Budman1758 What is the Point?

Budman1758, 2018-03-29:
Read the first sentence in my first post.

micropet, 2018-03-29:
Everything OK.
Thank you